qba667/FlySkyI6 #2: Compile Source

Pecacheu, 2017-03-30:
The source code is under FlySkyI6/source right? How do I compile it? The README mentions **arm-none-eabi**, which I know I have installed under
`AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/`
due to the Arduino IDE, but how do I use it?

Pecacheu, 2017-04-02:
Here's the Windows batch script I made:
<pre>
@echo off
setlocal EnableDelayedExpansion

set EABI_PATH="%LocalAppData%\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin"

@RD /S /Q temp

for /D %%d in (*) do (
	set DIRS=!DIRS! -I %%d
)

mkdir temp

echo Dir String: '%DIRS%'

for /D %%d in (*) do (
	echo ---- Compiling Directory: %%d
	rem C Files:
	for %%f in (%%d/*.c) do (
		"%EABI_PATH%/arm-none-eabi-gcc" %DIRS% -c -mcpu=cortex-m0plus -g "%%d/%%f" -o "temp/%%~nf.o" -mthumb -g3
		if !errorlevel! neq 0 (
			pause
			exit /b !errorlevel!
		)
	)
)

rem "%eabi_path%/arm-none-eabi-as" -o fsi6.o fsi6.asm -mcpu=cortex-m0plus -mthumb -g3
rem "%eabi_path%/arm-none-eabi-objcopy" -O binary fsi6.o firmware.hex

pause
</pre>
And this is what I get when I run it:
<pre>
Dir String: ' -I drivers -I fonts -I source'
---- Compiling Directory: drivers
In file included from drivers/fsl_common.h:38:0,
                 from drivers/clock_config.h:34,
                 from drivers/clock_config.c:68:
drivers/fsl_device_registers.h:52:6: error: #error "No valid CPU defined!"
     #error "No valid CPU defined!"
      ^
In file included from drivers/fsl_common.h:126:0,
                 from drivers/clock_config.h:34,
                 from drivers/clock_config.c:68:
drivers/fsl_clock.h:309:28: error: 'MCG_C2_EREFS0_MASK' undeclared here (not in a function)
     kOSC_ModeOscLowPower = MCG_C2_EREFS0_MASK, /*!< Oscillator low power. */
                            ^
drivers/fsl_clock.h:324:28: error: 'MCG_C2_HGO0_MASK' undeclared here (not in a function)
                            MCG_C2_HGO0_MASK, /*!< Oscillator high gain. */
                            ^
drivers/fsl_clock.h:331:18: error: 'OSC_CR_SC2P_MASK' undeclared here (not in a function)
     kOSC_Cap2P = OSC_CR_SC2P_MASK,  /*!< 2  pF capacitor load */
                  ^

< ... You Get The Idea ... >

drivers/fsl_clock.h: At top level:
drivers/fsl_clock.h:945:43: error: unknown type name 'OSC_Type'
 static inline void OSC_SetExtRefClkConfig(OSC_Type *base, oscer_config_t const *config)
                                           ^
drivers/fsl_clock.h:971:35: error: unknown type name 'OSC_Type'
 static inline void OSC_SetCapLoad(OSC_Type *base, uint8_t capLoad)
                                   ^
In file included from drivers/clock_config.h:34:0,
                 from drivers/clock_config.c:68:
drivers/fsl_common.h:181:30: error: unknown type name 'IRQn_Type'
 static inline void EnableIRQ(IRQn_Type interrupt)
                              ^
drivers/fsl_common.h:198:31: error: unknown type name 'IRQn_Type'
 static inline void DisableIRQ(IRQn_Type interrupt)
                               ^
drivers/fsl_common.h:246:24: error: unknown type name 'IRQn_Type'
 void InstallIRQHandler(IRQn_Type irq, uint32_t irqHandler);
                        ^
drivers/clock_config.c: In function 'CLOCK_CONFIG_SetFllExtRefDiv':
drivers/clock_config.c:97:5: error: 'MCG' undeclared (first use in this function)
     MCG->C1 = ((MCG->C1 & ~MCG_C1_FRDIV_MASK) | MCG_C1_FRDIV(frdiv));
     ^
drivers/clock_config.c:97:28: error: 'MCG_C1_FRDIV_MASK' undeclared (first use in this function)
     MCG->C1 = ((MCG->C1 & ~MCG_C1_FRDIV_MASK) | MCG_C1_FRDIV(frdiv));
                            ^
Press any key to continue . . .
</pre>

qba667, 2017-04-02:
I was using Freescale Kinetics the command used for compilation:

`arm-none-eabi-gcc -mcpu=cortex-m0plus -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -Wall  -g -D"CPU_MKL16Z64VLH4" -I../startup -I../board -I../utilities -I../CMSIS -I../drivers -std=gnu99 -MMD -MP -MF"source/file.d" -MT"source/file.o" -c -o "source/file.o" "../source/file.c"`
Keep in mind that is not complete firmware. After compilation original file needs to be patched.

qba667, 2017-11-10:
@Pecacheu 
Try make script from source/build.zip
because the size of blocks depends on compiler optimization some sections must be optimized.

closed 2017-11-23 by qba667