qba667/FlySkyI6 #3: Find an error about FlySkyI6

lossday, 2017-07-07:
Hello.
Channel 6 is set at VRB in the helicopter variable pitch type, and the correct position is the throttle rocker.
Why not support IA6 and support only IA6B?

Can you modify it? Where can I find the source code?

qba667, 2017-07-07:
@lossday please provide more details - at least affected version.
I believe I have fixed this one long time ago please see:
https://github.com/benb0jangles/FlySky-i6-Mod-/issues/64
What you mean by not supporting IA6? Is it working as expected - IA6? can send 14 channels over "semi" IBUS.
Source code is located in source directory.

lossday, 2017-07-07:
Thank you very much. I saw it, too
 i couldnt bind my receiver
ia6     is  not  ia6B

qba667, 2017-07-07:
@lossday  there was a bug in binding try to flash with one of the updaters:
https://github.com/qba667/FlySkyI6/tree/master/release
It will fix binding issue. Is the variable pitch working now correct?
IA6 has semi IBUS (almost the same format) - the signal can be routed to one of the pins of RX, it is already supported in Cleanflight and others.

lossday, 2017-07-07:
It is now working fine, only found a small problem at the end point set, but I believe it will not be a big problem, I will test it

Again thank you so much

qba667, 2017-07-07:
I am not sure if endpoints are considered at all in this case.

closed 2017-07-07 by qba667