qba667/FlySkyI6 #4: Feature Request: Ability to calibrate Ext. Voltage

MarMed, 2017-07-09:
I'd would really like a way to adjust the read voltage from a FS-CVT01 as they are notorious for being off by .2V+

Thanks!

qba667, 2017-07-09:
@MarMed I have experimented a little by kidnapping telemetry method. It can be done - I have to add calibration menu. Maybe in few days.

wsqzdz, 2017-07-10:
when will  the  flyplus1.6  come？how  to use  the  channels 11-14，and  why  only have  ppm 4 5 6，123？

wsqzdz, 2017-07-10:
FlyPlus 1.7.。。

qba667, 2017-07-10:
PPM is usually representing 8 channels - the radio is parsing channels channels 7,8 incorrectly and apply them to channels 1,2 - this results in random jump of values. PPM input handling must be improved first.
Go to Extra menu, Aux Channels - configuration can be done there.

MarMed, 2017-07-10:
@qba667 Thank you for your consideration!

wazoo666, 2017-11-29:
@qba667  
hi.  I am making a different mod for IA10b receiver to monitor battery voltage, just using BIND pin as input to battery voltage metering channel on uC (PA1)  (cutting vcc wire and removing R4 to make direct connection to input)
I am using the naza basic pmu (not V2) that provides proportional voltage output.
Now I can read from flysky FS-i6x (firm v1.0 30/maz/2017) voltages from 4,03v (10,000v battery input volgage) to 7,739v (18,000 Vin) 
It is posible to add and offset value or calibrate option to RX battery on firmware as i6x do with its own batteries ?
Thanks.  good job.

qba667, 2017-11-29:
I know that there is build in calibration in X6B and i6B, it looks it can be configured with use of configuration packet or default value from uP is being used if nothing provided. 
The formula looks like:
adcValue * configuredValue/1000;
Maybe I will be able to find way to configure it from TX.

wazoo666, 2017-11-29:
@qba667 
oooohhh , it will be great!!   thanks for your fast reply.
then mod to ia10b will be more easy.
recently I have post a message on RCgroup showing my mod adding some pictures.
I hope you can see them.

[image]
[image]

qba667, 2017-11-29:
@wazoo666 I believe it can be calibrated. What are voltage divider values?

wazoo666, 2017-11-29:
in my mod there is no "voltage divider" just a direct input from naza pmu to ia10b input (with R8 and C14 in parallel but probably do not affect a lot in measurement) (just not to desolder  :-D )
with a variable power supply I show that pmu provides 0,888v (10,000v real) ..... to 1,597v (18,000v real)
because pmu is designed from 2s to 6s and the input on uC must not be exceed 3,3v
applying the same rule pmu will provide abou 2,24v for its maximum 6s

wsqzdz, 2017-12-02:
would you plan to make a simple  osd？

wazoo666, 2017-12-03:
yes, an easy way to know RX battery voltage without external devices.  :-D

qba667, 2017-12-11:
@wsqzdz 
The calibration frames can be indeed send - so far I found it for IA6B. It is far to complex to handle it in I6 - so calibration on TX level will be added.

qba667, 2018-01-08:
@wazoo666 @MarMed  Calibration feature added please check:
https://github.com/qba667/FlySkyI6/releases/tag/1.7.3

wazoo666, 2018-01-08:
@qba667 

ThankU very much.  I will check as soon as posible.
Can I download original firmware to make as backup ?
Because original firmware does not apear on FlySky website

wazoo666, 2018-01-09:
Oooopsss

Can not update.  I have discovered that Simulator/Trainer port does not have a Serial port in this i6x version, like in i6 version.
Once I openned it , PCB board shows D+ and D- , that is suspicious to be USB , and following tracks I have discovered that they goes to CPU pins 70/71 ( datasheet say USB_DM/USB_DP)
Once I wired with a pure USB cable, computer detect new HID device , not COM port.
Now I can "update" with genuine i6x updater from FlySky , but not with modded COM Updaters.

Bad luck for me.
**@qba667**  I can not enjoy your hard work.

[image]

qba667, 2018-01-09:
@wazoo666 firmware is only for I6, it was never intended for I6X the code base is the same - it could be ported but this is not work for me.

wazoo666, 2018-01-10:
OK  @qba667  I understand that it is not your job , probably someone can port to I6X.
I will wait for that.
Anyway, I want to thank you for your work and time dedicated to the community.

qba667, 2018-01-10:
@wazoo666 please create complete dump of firmware -> as described here:
http://fishpepper.de/projects/openground/
the MCU should have same boot-loader. 
We will see what can be  done.

qba667, 2018-01-19:
@wazoo666  please see:
https://www.rcgroups.com/forums/showthread.php?3020949-FlyPlus-firmware-for-FlySky-I6X#post39015734

closed 2018-01-19 by qba667