qba667/FlySkyI6 #5: Quad icon !

yo8aiv, 2017-07-10:
Would be nice to have an quadcopter icon instead of plane.
Thx.

jordirafel, 2017-07-14:
+1 or a custom load image 💃

jordirafel, 2017-07-14:
[image]

qba667, 2017-07-15:
@jordirafel perfect! I will add this one in place of current one!

qba667, 2017-07-15:
Or maybe I will write small app to change the icon to custom one.

yo8aiv, 2017-07-18:
great !

qba667, 2017-07-28:
@yo8aiv, @jordirafel I will add method to load custom PNG into updater so you can load whatever you want.

jordirafel, 2017-07-28:
Great! Many thanks

qba667, 2017-07-29:
@jordirafel, @yo8aiv GFX editor added:
https://github.com/qba667/FlySkyI6/releases/download/1.6.3/FW.gfx.editor.zip
Learn how to use it and write some documentation.

jordirafel, 2017-07-31:
This is awesome, its possible to change other icons too! Many thanks :)

yo8aiv, 2017-08-04:
thanks. in progress.

closed 2017-10-04 by qba667