qba667/FlySkyI6 #6: RSSI Output for RX (iA6B)

2black0, 2017-07-15:
can we get rssi output from rx ia6b?

qba667, 2017-07-17:
If you use an Error in channel assignment page. RSSI itself is hard to scale.

2black0, 2017-07-18:
Can we feed it to osd?

2black0, 2017-07-21:
can you share the wiring?

2black0, 2017-07-22:
RSSI (Error) send on channel 11 (in older version)

sorry just need confirmation, so we can get rssi output for osd in ch11? so just enable pwm rssi in betaflight and choose CH11? right?

qba667, 2017-07-28:
@2black0 just select RSSI channel in radio setup tab.

KnuckleUpFPV, 2017-08-09:
This works on any diversity antenna receiver by flysky. Set err from 10 channel mod or newer to an extra channel. I use channel 9. Then I go to betaflight receiver tab and change rssi to channel 9 and save. I have perfect rssi and its extremely accurate. I also have a 4w booster that I use. Unfortunately can only boost signal to the receiver and not the signal back to the tx. With booster off my rssi counts down from 100 to 0. If I hit 0 I failsafe. With my booster on I can hit 0 and keep going to an untested/verified distance. Ive gone a mile on stock tx. With booster on, I have gone 1.5 miles. Nervous to go further without the appropriate open space.

closed 2017-10-04 by qba667