qba667/FlySkyI6 #7: 6 position rotary switch E

matarum, 2017-07-17:
HI,

I was wondering if there is some kind of way to have a 6 position switch configured on switch E implementation. It could be useful in pixhawk projects.

Thanks

qba667, 2017-07-17:
@matarum unfortunately not but you can use SwC (can be configured in extra menu) as 6 position switch and them use 2 position switch in place of SwC.

matarum, 2017-07-18:
Sorry, I'm not sure I understand.
i can replace the 3 position switch with a 6 position switch on SwC (and configure it as variable in extra menu). Then I could replace 2 position switch with 3 position switch I removed from SwC on SwB and then put that 2 position switch on SwE?
is it right?

qba667, 2017-07-20:
@matarum, yes this is absolutely right.

matarum, 2017-07-20:
could I use a potentiometer instead of 6 pos switch on SwC?

matarum, 2017-07-22:
ok. got it.

closed 2017-07-22 by matarum