qba667/FlySkyI6 #10: FS-i6 with TGY-CAT01

cookieqk, 2017-07-27:
Hi,
Looks like there's ZERO owners & users of turnigy's altitude sensor... 
 I've connected it to FS-iA6b receiver.  On sensors list in my FS-i6 with he last MOD (14_channels_SWE) there's just a "Pressure sensor". It shows 9 digits on the screen e.g 367123788. There's no possibility to change unit (Pa, psi, bar... ) or to do anything else with it.

From hobbyking: "Reliably monitor the altitude of your model with the TGY-CAT01 Altitude Sensor for use with the TGY-iA10 receiver and TGY-i6 transmitter. Use it to measure the real-time altitude of your drone, plane or heli directly on your Turnigy i10 transmitter."

I even updated my FS with Turnigy's TGY-i6 firmware, but it still hadn't work correct.

Is it possible, that connecting CAT01 to FS(TGY)-iA10(B?) receiver would change anything?

qba667, 2017-07-28:
@cookieqk there is typo in HobbyKing description - I have guessed the format of the sensor because I don't have one. There is code base prepared to handle the sensor but it is not fully enabled. Maybe I will enable it in future - depending on resources.

cookieqk, 2017-07-28:
@qba667 = Jakub => POL? :-)

qba667, 2017-07-28:
@cookieqk your reasoning seems to be correct;)

qba667, 2017-08-08:
@cookieqk  the CAT01 damaged my IA6B. But I get some log entries:
[image]

cookieqk, 2017-08-09:
This is not very good news :-/ My iA6b works fine with cat01 - theoretically...

JakubMichal, 2017-09-08:
Hi. Anything new about this Cat01?

qba667, 2017-09-11:
@JakubMichal 
it is almost working some small improvements are needed.

JakubMichal, 2017-09-11:
You are the best! ;-)

qba667, 2017-12-11:
@JakubMichal @cookieqk
It is done! It took a lot of flash space - but seems to be working. It could be used as ASL and relative value. 
https://github.com/qba667/FlySkyI6/releases/tag/1.7.1
Have a lot of fun!
@cookieqk thank you very much for your support.  
Wysokich lotów panowie

qba667, 2017-12-11:
@JakubMichal @cookieqk 
As bonus you can use vario functionality to get notifications about value change.

JakubMichal, 2017-12-11:
Super! Thank you very much! ;-)

qba667, 2017-12-15:
No complains  == working solution. Closing.

closed 2017-12-15 by qba667