qba667/FlySkyI6 #11: Smartaudio

lrusnac, 2017-08-01:
Just an idea that we could maybe investigate? It would be cool to have a smartaudio controller, in particular being able to set a VTX channel and band from the TX before powering up the RX.

more info: https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio

yo8aiv, 2017-08-04:
+1 on this.
i have a VTX like that on the way.

Yenya, 2020-05-12:
@lrusnac, @yo8aiv: what would this feature mean for the Tx firmware? Sorry, I don't fly FPV copters myself. I have found the following page:

http://www.multirotorguide.com/guide/guide-smartaudio-vtx-control-and-how-to-set-it-up/

where they explain that SmartAudio can be configured via Betaflight flight controller using the ordinary transmitter sticks/channels movement. What does it have to do with the FlySky firmware?

Thanks in advance for the explanation.