qba667/FlySkyI6 #12: new X6B - not binding !

yo8aiv, 2017-08-04:
hello, anyone had this problem >>??

new X6B - not binding ! tried ppm and ibus and nothing 
receiver is blinking but binding not happening.
Thanks.

yo8aiv, 2017-08-04:
after further tests i managed to bind the X6B this way:
- supply from a rs232/ftdi adapter 5V 
- going into the secret menu, see here 
[image]

its weird that it didn't work the usual binding way - with the bind key - is that a known bug ??!!

qba667, 2017-08-04:
@yo8aiv are you using FW from my (this) repository? Older version has a bug related to binding?

yo8aiv, 2017-08-04:
Thanks for the fast answer.
Well if i remember well primarily i think i downloaded this firmware from rcgroups page not from here directly.
after i have found the repository -- wasnt easy .. and downloaded this one ,, but i'm not shure i flashed this latest version.
i'll try reflashing this one and see how it behaves and i'll get back 

this one i 've flashed https://www.rcgroups.com/forums/showatt.php?attachmentid=10192248&d=1499865537

qba667, 2017-08-04:
The bug was known in this version - use latest one from repository.

yo8aiv, 2017-08-04:
OK. Thanks.
Flashing.

yo8aiv, 2017-08-04:
Done. Works.
Its nice you've added information on the firmware page regarding moded version. 
Cheers. see my other proposal about flashing alarm on screen text.
Thx again.

closed 2017-08-05 by yo8aiv