qba667/FlySkyI6 #15: make GFX editor multiplatform and share the sources

sl1pkn07, 2017-08-11:
Hi

is possible make the GFX editor multiplatform?

where can download the sources?

not all people use windows system ;)

greetings

qba667, 2017-08-14:
@sl1pkn07 
The editor has been written in c# if mono is fine for you I can provide the source.

sl1pkn07, 2017-08-14:
mono can work natively on linux. but i'm not sure if the code of gfx editor is windows centric

qba667, 2017-08-14:
It is not:) you will need just one method to create byte array from bitmap.

qba667, 2017-08-14:
@sl1pkn07 here you have the source:
https://github.com/qba667/FlySkyI6/tree/master/GFXeditor

sl1pkn07, 2017-08-14:
tnx!

sl1pkn07, 2017-08-16:
what you mean with this?

> It is not:) you will need just one method to create byte array from bitmap.

anyway. i can compile the code with mono 5.0.0 (provided by package distro) with `xbuild`

greetings

qba667, 2017-08-17:
I meant  the gfx editor is not windows centric. 
Great, you can share the binary.

sl1pkn07, 2017-08-17:
done 

https://github.com/sl1pkn07/FlySkyI6/commit/97da338baccb7a01f82092bb33a557d6b11cff0e

sl1pkn07, 2017-09-22:
update https://github.com/sl1pkn07/FlySkyI6/commit/44b590b118f60a65d711518fae9db3488c30615e

NOTE: for only x86_64

closed 2017-10-27 by qba667