qba667/FlySkyI6 #16: Endstop adjustment for channels 7 through 10

scruffy4386, 2017-08-14:
As my title suggest, there is no means to adjust the endstops for these added channels. I see youve put in additonal items in the extra menu. Is there room there for endstop adjustments?  Also, i seems to have a perstistant beep every few minutes that i cant figure out.

qba667, 2017-08-14:
@scruffy4386 Endpoint configuration can be stored - but some improvements os existing configuration page is needed.
Alarm can be triggered by TX voltage configuration or alarm settings check both options in extra menu - the memory is initialized first time with some random data.

kelvinpmr, 2017-08-28:
Hi everyone! i`m having the same inssue, the problem is that Naza Controller set the flight mode in channel 7 on PPM configuration, and without the end point and reverse configuration for ch7 is impossible to set the flight mode on a switch, now, i`m using a VrA switch, but this is not the right thing.

qba667, 2017-08-28:
@kelvinpmr it is possible check file:
https://github.com/qba667/FlySkyI6/blob/master/source/source/alt.c
 There you can find method getAuxChannel endpoint values must be provided as parameters and included in calculation. 
I am in business trip for some time now - but maybe after I will be back I will check how it could be handled correctly.

kelvinpmr, 2017-08-28:
Wow! Thx for your response! I'm seen some people with the same problem, basically everyone How have the dji naza controller André uses ppm, have this problem. Im not realy a good programmer and i'm on road now too. But o Will try something to this values. Thx!

qba667, 2017-08-28:
@kelvinpmr the modConfig2.modelConfig[modelIndex] can be used to store endpoints but it probably needs to me moved somewhere else in memory because we need about 4 bytes for every channel.

kelvinpmr, 2017-08-28:
I don't Read the code yet, but, the same string of the endpoint configuration on ch1 to ch6, can't bem used to ch7 - 10 configuration too? What the difficult to put this to work until now? This is a problem that exists last than the first firmware modded version right?

qba667, 2017-08-28:
@kelvinpmr here is the disassembly:
https://github.com/benb0jangles/FlySky-i6-Mod-/issues/58

kelvinpmr, 2017-09-18:
Hi, @qba667 some evolution on channels endpoint?