qba667/FlySkyI6 #17: share the changes of you flysky-updater

sl1pkn07, 2017-08-16:
Hi

since the flasher updater (flysky-updater) provided by @mhils repositoy is writen in GO (and can build in lot platforms with lot platforms (cross-compiling)) , and you make changes in it

you can provide/upload the changes?

greetings

mhils, 2017-08-16:
Haha. Thanks for tagging me - I wasn't even aware of this very cool project here. 😄 
Thanks @qba667 for making the updater so awesome! Also thanks for prominently giving attribution, that certainly wouldn't have been required. ☺️ 

I licensed it under the MIT license, so you do not have to share any changes of course. Nonetheless, I'll second @sl1pkn07's request as I'd be curious myself! 😃

qba667, 2017-08-17:
@sl1pkn07 repository can be found here:
https://github.com/qba667/flysky-updater/blob/master/flysky_updater.go
@mhils you have done most difficult work including reversing and writing documentation. 
There was only one bug in writing method - related to file size.
Then I have added some cosmetic - also CRC can be verified (currently part of my python script). 
Implementation is not perfect because I am not Python or GO programmer, even I am not ARM assembly or embedded system engineer ;)

sl1pkn07, 2017-08-17:
tnks

```
goroutine 1 [running]:
main.main()
        /home/sl1pkn07/aplicaciones/flysky-updater-qba667/flysky_updater.go:210 +0x1de9
```

when launch without arguments (linux)

qba667, 2017-08-17:
@sl1pkn07, try replacing 
```c
var portStr string = fmt.Sprintf("COM%d", p)
```
with 
```c
var portStr string = fmt.Sprintf("/dev/ttyS%d", p)
```

sl1pkn07, 2017-08-17:
same output

qba667, 2017-08-17:
The line 210 indicates:
panic("No firmware found!!!") 
Add firmware file to directory of updater.

sl1pkn07, 2017-08-17:
now works. but need tweaks

~~~patch
diff --git a/flysky_updater.go b/flysky_updater.go
index ec4d906..5dfe358 100644
--- a/flysky_updater.go
+++ b/flysky_updater.go
@@ -212,7 +212,7 @@ func main() {
 
 	var comports []int;
 	for p := 0; p < 24; p += 1 {
-		var portStr string = fmt.Sprintf("COM%d", p)
+		var portStr string = fmt.Sprintf("/dev/ttyUSB%d", p)
 		c := &serial.Config{Name: portStr, Baud: 115200, ReadTimeout: time.Second * 1}
 		s, err := serial.OpenPort(c)
 		if err != nil {
@@ -232,13 +232,13 @@ func main() {
 		portIndex = comports[0];
 	} else{
 		for p := 0; p < len(comports); p += 1 {
-			fmt.Println(fmt.Sprintf("[%d] COM%d", comports[p], comports[p]));
+			fmt.Println(fmt.Sprintf("[%d] /dev/ttyUSB%d", comports[p], comports[p]));
 		}
 		fmt.Print("Please select serial port: ");
 		fmt.Scan(&portIndex)
 	}
 
-	fmt.Println(fmt.Sprintf("Selected serial port: COM %d", portIndex));
+	fmt.Println(fmt.Sprintf("Selected serial port: /dev/ttyUSB%d", portIndex));
 	
 	var fwNames []string;
 	var firmwareFile int;
@@ -298,7 +298,7 @@ func main() {
 		continue;
 	}
 
-	c := &serial.Config{Name: fmt.Sprintf("COM%d", portIndex), Baud: 115200, ReadTimeout: time.Second * 1}
+	c := &serial.Config{Name: fmt.Sprintf("/dev/ttyUSB%d", portIndex), Baud: 115200, ReadTimeout: time.Second * 1}
 	s, err := serial.OpenPort(c)
 	if err != nil {
 		fmt.Println(err);
~~~

mhils, 2017-08-18:
@sl1pkn07: I can hack something together that incorporates @qba667's improvements and fixes this :)

mhils, 2017-08-18:
> There was only one bug in writing method - related to file size.

@qba667: Care to elaborate? Are you referring to [this](https://github.com/qba667/flysky-updater/commit/e25dfd182c5935e57d93ac973695d70dd68f2893#diff-637602cfdd9faa867bb78f1209bccd0aR287) change where you cut off the first 0x1800 bytes if the total size exceeds 0xE7FF?

mhils, 2017-08-18:
@sl1pkn07: I added some improvements to the editor and fixed your bug, would you mind taking a look at  https://github.com/mhils/flysky-updater/blob/master/flysky_updater.go and testing it? 😃

qba667, 2017-08-18:
@mhils I just remember that in initial version I was getting strange errors when file length was not multiple of 256. Possibly this bug is still present. Maybe I fixed it in released version - hard to say because I was working on it somewhere else and I have no access to this machine right now. It was also related to progress bar.

mhils, 2017-08-18:
Ah, that's helpful, I think I found a flaw. Let me see!

qba667, 2017-08-18:
@mhils you should check if this is last chunk and reduce end offset. After writing last chuck you should break the loop.

mhils, 2017-08-18:
Ha - thanks for the great pointer. I fixed the updater to work with firmware that's not kilobyte-aligned and also added a quite nice selection prompt for both serial port and firmware image. 😄 

https://github.com/mhils/flysky-updater/releases/tag/v2.0

@qba667: I also added you as a collaborator to the repo. Please feel free to perform changes as you wish. If there's anything else that'd be useful to you, let me know, and I'm happy to help ☺️

qba667, 2017-08-18:
@mhils Thank you very much! Very nice implementation. I will use the updater with next version of FW, there is still few topics outgoing. In free time I will incorporate checksum checking/calculation.
Great work! 
Regards from place located not so far from Innsbruck;)

sl1pkn07, 2017-08-18:
@mhils done

a little thing

```
? Serial Port: 
No serial port selected: interrupted.┌─┤[$]|[sl1pkn07]|[sL1pKn07]|[~/aplicaciones/flysky-updater]|
```

~~~patch
diff --git a/flysky_updater.go b/flysky_updater.go
index 9bf60e2..06b6f94 100644
--- a/flysky_updater.go
+++ b/flysky_updater.go
@@ -229,7 +229,7 @@ func choose(item string, options []string) (string, error) {
        }
 
        if ret == "" {
-               return "", errors.New("interrupted")
+               return "", errors.New("interrupted\n")
        }
        return ret, nil
 }
~~~
(?), sorry, I'm not a coder

thanks to both for the work!

mhils, 2017-08-18:
You are missing a newline after the error you mean?

sl1pkn07, 2017-08-18:
yep, Is shown in any error

mhils, 2017-08-18:
Here you go: https://github.com/mhils/flysky-updater/releases/tag/v2.0.1

sl1pkn07, 2017-08-18:
done

sorry If i am abusive, testing some thigs, foud this:

when launch the updater without connected the serial dongle (also without connecting to the TX). and then connect it (still without connect to the TX), when introduce the port (dev/ttyUSB0) and hit the enter, show this:

```
└───╼  ./flysky-updater 
Using NO_SWE_updater_07_10_19_22.bin as firmware image.
Could not autodetect serial port. Please enter manually.
? Serial Port: /dev/ttyUSB0
? Serial Port: /dev/ttyUSB0
Cannot ping the remote: EOF.
``` 

but when connect the dongle (without connecting to the TX) before run the updater, show this:

```
└───╼  ./flysky-updater 
Using NO_SWE_updater_07_10_19_22.bin as firmware image.
Using /dev/ttyUSB0 as serial port.
? Start flashing? (y/N) 
```

This means that should always connect the dongle before launch the updater, right?

greetings

mhils, 2017-08-18:
> sorry If i am abusive, testing some thigs, foud this:

No worries - I appreciate your self reflection here, but I'm always happy for constructive feedback on my overengineered pet projects. :)

> This means that should always connect the dongle before launch the updater, right?

As a cautionary measure, you get a confirm prompt iff the updater autoselects the serial port, because I wanted to avoid that the updater sends commands to another serial device by accident before the user has a chance to react.

Not connecting the TX (but connecting the dongle) should ultimately always give you a `Cannot ping the remote: EOF.` (either after "serial port" or "start flashing" prompt).

Flashing should always succeed if you connect both dongle and TX at any point before confirming the "serial port" or "start flashing" prompt.

Does that make sense? :-)

sl1pkn07, 2017-08-18:
thanks for the clear explanation.

yep. now test with the same way as before, but now with the TX, have the same behavior, this time with/out enable the firmware update mode in the TX. 

all ok then

i think this thread is finished and can be cose. any objetions?

closed 2017-08-29 by sl1pkn07