qba667/FlySkyI6 #18: Failsafe

pad73, 2017-08-23:
Hallo, is it possible to add failsafe setting to all channels? It is for basic 6ch only.

yo8aiv, 2017-08-29:
+1 on this also.
I want to use/set ch14 for RSSI with failsafe on it if its possible. 
Thanks.

pad73, 2017-08-30:
OK, we are two people who need it :)

openmy, 2017-09-21:
ia6b ppm failsafe setting not work

qba667, 2017-11-13:
@openmy  for channels 1-6 should be fine.
@yo8aiv @pad73  can be done - but it will reduce model memory 
4*8 bytes  = 32 bytes for model - so about 3 models less in total

kelvinpmr, 2017-11-13:
Hi! Jakub! the inssue for the 7th channel endpoint problem has closed, but, we continue have problem with NAZA Users, can you just put the 7th channel end point configuration? you already said that this radio have a little space in memory for this thing but, only channel 7 can be done? thx!

qba667, 2017-11-13:
@kelvinpmr I was not aware of this issue - maybe even channels 7 and 8. But I need to change Dave's  assembly code. I will keep this one in mind!

kelvinpmr, 2017-11-13:
Hi @qba667, here the issue: https://github.com/qba667/FlySkyI6/issues/16

Thx a lot! i'm so glad to hear that! today Naza Users with PPM Connection like the FS-IA6B receptor uses, need to put the flight mode (GPS-ATTI-FailSafe) on a VR Switch, work, ok, but one little touch on knob and the flight mode can be changed and some crash can happen.

qba667, 2018-01-06:
@kelvinpmr endpoints added:
Without labels but it should be working.
Menu->System->Extra->Mix

Channel 07 [Min limit][Sub trim][Max limit]
Channel 08 [Min limit][Sub trim][Max limit]
Channel 09 [Min limit][Sub trim][Max limit]
Channel 10 [Min limit][Sub trim][Max limit]
Channel 11 [Min limit][Sub trim][Max limit]
Channel 12 [Min limit][Sub trim][Max limit]
Channel 13 [Min limit][Sub trim][Max limit]
Channel 14 [Min limit][Sub trim][Max limit]

kelvinpmr, 2018-01-06:
Hi! @qba667 thx so much for this implement this is so awsome! This Will help a Lot of users! Thx again!