qba667/FlySkyI6 #19: Feature: Display Backlight Time

DimaVIII, 2017-08-30:
Hey,

It will be one of the best features if there will be a option set the time of the display backlight.
I understand it might be a lot of work.
May instead:
Can you just publish a version where the backlight never turns off.

qba667, 2017-09-18:
@DimaVIII it should be 2 byte change, I will provide the offset to be changed.

DimaVIII, 2017-09-19:
@qba667 thank you! Looking forward to it!

qba667, 2017-11-13:
@DimaVIII I got working radio back - I will search for place to be changed.

qba667, 2017-11-13:
@DimaVIII 
Addressed in full image with boot-loader, if you are using updater image substrac 0x18
Timout is stored at the address 0x2BD8 -> 20 4E 00 00 = 4E20 Increase this one
Call turning of back light is located at 0x286E -> 03 F0 11 FA replace by nops C0 46 C0 46

closed 2018-02-08 by qba667