qba667/FlySkyI6 #22: Lipo-LiIon 2S - Cannot read voltage above 6.6V

yo8aiv, 2017-10-13:
Lipo-LiIon 2S - Cannot read voltage above 6.6V.

Is this a hardware limitation ?
i tried calibrating from advanced hidden menu but then the voltage scale goes wrong. so it's a dead end.

anybody experienced this? 
radio works fine with 2S LiIon cellphone baterries .. just unable to add an proper alarm since voltage read isnt working.

Thanks for your work.

qba667, 2017-10-13:
@yo8aiv  yes it is limited by values of resistance in voltage divider - you are lucky that ports are tolerant and allow voltage above 3.3V.  To make it working just change smd resistors. Using 2s battery is waiting of energy because after all it is linear covered to 3.3 - so you have an heater:)

yo8aiv, 2017-10-14:
thanks for the answer.
i'll do.

closed 2017-10-27 by qba667