qba667/FlySkyI6 #32: model memory

gazeddy, 2017-11-24:
is it possible to recover the model memory before i flash. i have a couple of multirotors with depinned ia6b receivers that are encased in epoxy with only three wires coming out. so have no way of rebinding them without risking damage.

qba667, 2017-11-24:
@gazeddy 
Theoretically we can send them over UART. But to do it you need to flash custom firmware:)
The other way is read I2C eeprom. I have use method described here:
https://www.richud.com/wiki/Rasberry_Pi_I2C_EEPROM_Program

qba667, 2017-11-24:
@gazeddy depending how old current firmware is:)

gazeddy, 2017-11-24:
as far as im aware its the original version so not holding my breath on it. thanks for your help

closed 2017-11-24 by gazeddy