qba667/FlySkyI6 #34: flysky i6x

jmvau54, 2017-11-27:
Hi
I have a question that I don't find answser

Is your firmware compatible with flysky i6x ? or not ?

Thank you for your answer.

lrusnac, 2017-11-27:
@jmvau54 it is not compatible, the i6x has different hardware from i6

qba667, 2017-11-27:
@lrusnac that is true, but firmware is similar same rules, similar methods. 
Lately my code evolved a little so maybe it could be code base for I6X as well. Someone must invest time in disassembly, personally I like STM much more than Freescale. There is much more resources - things are just easier. Working on next device would be much simpler. However the best solution would be to rewrite whole firmware. We know protocol or some parts of protocol there is still a lot of magic that is present in IA6B which I am not able to understand. At lest approach can be applied - maybe someone one day will try.

lrusnac, 2017-11-27:
@qba667 that's interesting, I would like to see a port for the i6x, but I think the answer to his question is "No" because you can't just upload the current firmware to the i6x

jmvau54, 2017-11-29:
Hi
Thanks
Do you know an alternative firmware for i6x ?
I have a th9x with er9x. With it, I can do all I want. 
I have bought an i6x to use wirh boat, car and digger. But programming is very poor.
I would buy an i6.

qba667, 2017-11-29:
@jmvau54 no one that I aware of - they were rumors about OpenTX. FlySky contacted bsongis from open TX, they promise resources and sources but no feedback so far.

closed 2017-12-11 by qba667

qba667, 2018-01-23:
@jmvau54 We decided to create port for I6X. Please see:
https://gogetfunding.com/flyplus-firmware-for-flysky-i6x/

jmvau54, 2018-01-23:
Great news 
thank you