qba667/FlySkyI6 #37: Model 17..20 unselectable

koevoetsfjm, 2017-12-18:
After installing 1.71 i'm unable to select any model on position 17,18,19 or 20.

However, it's still possible to copy the models from 17..20 to positions 1..16 and move models from 1..16 to 17..20. 
It's just that you can't select them.

qba667, 2017-12-18:
@koevoetsfjm all additional functionality used model memory from 17-20 (in next version 15 to 20)
I have intentionally disabled model selection - but I have missed copying.  To be fixed in next version.

koevoetsfjm, 2017-12-18:
Scrapping for storage bytes... the amount of features you keep adding still amazes me and I hereby offer you my 15+16 model memory for even more features !

👍 Keep up the good work!

handstek, 2017-12-20:
having trouble with the extra's menu - alarm  when set to none and save once radio is turn off/on reverts back to a jumble

qba667, 2018-01-04:
@handstek I have wrote a reply in other ticket - as soon as I will find why it happens I will let you know.

qba667, 2018-01-06:
@handstek check against 1.7.2 - should be fixed - wrong memcpy variant used.
@koevoetsfjm models 15-16 taken:) Endpoint and subtrim implemented - check against 1.7.2

closed 2018-02-08 by qba667