qba667/FlySkyI6 #38: Image change

koevoetsfjm, 2017-12-18:
If you replace the first image (THE AFHDS2A image, which looks garbled in the graphics editor) you will brick your FS-i6 after upgrade. The display stays blank and the backlight turns on / off. I think it's due to the timer beeing displayed in that position. 

I don't know if that's fixable or not, but might be wise to add it to the readme.

I was able to fix it by holding the left-hand trimbuttons to right/down and turn the radio on, then re-flashing the untouched 1.71.

qba667, 2017-12-18:
As you probably seen there is no more AFHDS logo any more in the firmware. The space have been used for storing code. GFX editor has hard-coded offsets of images - and allows selection - but user is responsible for changes - because address can be provided so I don't care. 
So please do not change something that doesn't look like a image.

koevoetsfjm, 2017-12-18:
Yup, i noticed. It's no problem, but maybe worth a note ;-)

thx for your efforts, i'm really enjoying this firmware!

closed 2017-12-18 by koevoetsfjm

qba667, 2017-12-18:
@koevoetsfjm thanks, there is one more (probably last) update coming allowing sub-trim and endpoints on all channels.