qba667/FlySkyI6 #42: 1.7.2 version reads 1.7.1

unitware, 2018-01-08:
Downloaded and flashed https://github.com/qba667/FlySkyI6/releases/download/1.7.2/1.7.2.zip (the non-swe variant) but when I look at the version page it reads 1.7.1.

[image]

And I did not find the endpoint and subtrim in the system->extras (if that is the proper place)

qba667, 2018-01-08:
@unitware know the issue, I have just forgotten to increment the version number. Endpoint and subtrim can be set in Extra menu the menu entry is called Mix. Probably late entry in Extra menu.

qba667, 2018-01-08:
@unitware please check 1.7.3 few small fixes added.

unitware, 2018-01-08:
Ah, ok thanks. The "Mix" menu name confused me. 
I can try to write some documentation and take some screenshots of the "extra menu" and the functions if you want, or is that work already under way? (I noticed an empty file "Features.md" in the doc folder)

closed 2018-01-08 by unitware

qba667, 2018-01-08:
@unitware would be great, I was not able to take any photos because my backlight in the unit is broken. An other fully working unit I have is currently at home town. If you have time to do it it would help a lot. Were you able to find all additional features?

unitware, 2018-01-08:
I'll start a PR with some photos and then do my best guesses of the function and you can correct me?

By the way, will this firmware work in the i6x and i6s transceivers also?

Skickat från min iPhone

unitware, 2018-01-09:
Started to doc some stuff here, will add pictures and more text later on.
https://github.com/unitware/FlySkyI6/blob/doc-feature-ppm/doc/Features.md

unitware, 2018-01-09:
Now I noticed that you have begun to write a wiki page on github, It would make more sense to update that one of course. Let's treat my text as a draft and primary a placeholder for pictures and then you can move it to the wiki?

qba667, 2018-01-09:
@unitware I have added you as contributor. Do you can move your content to wiki. I will comment on your questions later.