qba667/FlySkyI6 #43: Double Flash with a little HW change

stefan73, 2018-01-10:
Looking at the PCB of the FS i6, I see a 24c128 flash SMD chip (U3). Pins A0,1,2 wired to ground.
It should be easy to solder the same chip on top of the existing one. Connect all pins to the underlying original but not A1. Connect a wire to A1 and connect this to +3.3V.
This should double the amount of flash available. We should see an additional I2C EEProm at 0x52, whereas the original flash remains at 0x50.

Do we want to try this?

qba667, 2018-01-10:
@stefan73 
You are speaking about EEPROM, there is enough memory there.
They are 2 different problems used MKL16Z64 has only 64 kBytes, there is MKL16Z128 but normal users will not be able to replace it, another limit is RAM limit but it should not be considered right now.
Other question is what more features do we need.

stefan73, 2018-01-10:
Thanks. I was thinking about the model memory. 
This seems to drop as you add features. So why not move the move the model memory to somewhere else. Soldering a flash with 8 pins on top seems easy enough for basically anybody. If there is still enough space in the EEProm this might not be needed.

qba667, 2018-01-19:
@stefan73 Over all at the end we have to load model memory to RAM.
There is enough space on EEPROM. The whole configuration is repeated number of times.

unitware, 2018-01-19:
Can it be used for rx-firmware-update over the air?

Skickat från min iPhone

qba667, 2018-01-19:
In my opinion no, because firmware update is implemented only in I10 there is no memory dump available for this unit so I have no idea how it is working. 
Reversing code from RX ix difficult. Theatrically I know what are typed of frames so It can be done. But without working system it will take me ages until I figure out what me be send. Other way would be attaching logic analyser to SPI of I10. 
If we ever discover protocol of update we can create custom firmware mentioned only for update - update receivers and revert standard firmware - this is the only way I see.

stefan73, 2018-01-20:
@qba667 Is it really implemented that all models are in RAM in parallel? -  I do not really know how much control over the frontend code you have. I would suggest to load only the current model selected to RAM and leave the remainder in the EEPROM. By that way EEPROM space directly translates into more models and the RAM situation relaxes.

qba667, 2018-01-20:
@stefan73 you are right it is very bad design. All model configuration is loaded at startup. There is also index of current model. So we can trick the radio load only single model, keep the index all the time equals 0. So without changing all the code (and searching where model index is being used) it can just work. We have to change also methods for model copying and similar. This is the work to be done on assembly level.

stefan73, 2018-01-22:
Bad design by the original SW indeed. I guess you could just set the # of models to 1 for the original code. Then "just" select model and copy model would have to be overloaded.
Anyways I will close the issue here and hope that the I6s approach leads to success.

closed 2018-01-22 by stefan73