qba667/FlySkyI6 #44: Extra->Mix menu

unitware, 2018-01-10:
Hi, excellent work!

As a new user i made a reflection on the mix menu.

1. I think it should be renamed to something like "Subtrim/Endpoints channel 7-14", or maybe "Trim/Endpoint" or similar. Mix I think is something different - that channels are mixed together.

2. "Default values are set to 0 - because of mod memory initialization. In such state normal mixing is being performed (min = -100%, subtrim = 0, max = 100%); Min, max and subtrim can be positive or negative there is no restriction - the only restriction is: Min < subtrim <max"

Could it be made so that if this state is detected (uninitialized module) then the values ar changed to the default values (-100%, 0%, 100%) and viewed like so?

qba667, 2018-01-10:
@unitware in 1.7.5 I have renamed the entry to end points. I will check how much space do we have maybe valid initialisation can be performed.

qba667, 2018-01-10:
Try this one initialization should be correct.
[fs-i6_updater_01_10_20_25.zip](https://github.com/qba667/FlySkyI6/files/1620168/fs-i6_updater_01_10_20_25.zip)

unitware, 2018-01-10:
Looking great!

unitware, 2018-01-10:
..I guess there is a reason why these are system wide and not in the model setup endpoints / subtrim menus along with the other channels?

qba667, 2018-01-10:
No they are not system wide those settings should be working for models. 
Setting are not aggregated together because because editing existing method must include extend functionality, keep old one and still use no more space then old implementation. Because logic becomes more difficult it is hard to fit it together. New implementation (smaller) can be fitted into free flash space.

unitware, 2018-01-10:
I see, thanks

qba667, 2018-01-10:
But maybe if we take space used by both methods unification will be possible.

unitware, 2018-01-10:
I'd rather use the space for other features, but then again clean implementation will always be welcome :)

kelvinpmr, 2018-01-15:
Hi @qba667 this weekend i try to use the endpoints feature, but does not work i try to set the Channel 7 endpoint but even going up and down in the number the cursor on the drone controller software don't move, There are something different the i need to do? Maybe change the value and save tô change the value on software? Thx!

qba667, 2018-01-15:
@kelvinpmr first you need to configure values then long press CANCEL to save. 
Since that point of time new values are being used to calculate channel value.

unitware, 2018-01-15:
Ok, I think this is different from the normal (ch 1-6) endpoint adjustments
- they can be adjusted live, which is better if possible. 

Skickat från min iPhone

Yenya, 2020-05-12:
@unitware, @kelvinpmr: if I read the issue correctly, the only thing left to do in this issue is to allow live adjustment of the endpoints/subtrims for channel 7 to 14 instead of having the changes visible only after saving the values. Is it correct?

If so, I would close this issue and fill a new one with more narrow subject.

Thanks,

-Yenya