qba667/FlySkyI6 #45: Can the X6B receiver use the combined ibus/telemetry?

unitware, 2018-01-10:
I have got a few "Flysky X6B 2.4G 6CH i-BUS PPM PWM Receiver for AFHDS i10 i6s i6 i6x i4x Transmitter"

They are good in that there is mounting holes that match the flight controller and they can monitor the battery voltage with dedicated input.

However there is no ibus telemetry pin, can these receivers be programmed with the i6a firmware?

qba667, 2018-01-10:
@unitware the hardware is almost the same - have never tried - go ahead be first one:)
Probably you will lose external voltage sensing - but if we have enough luck we can use version of IA6B I have patched - where telemetry and channel data is being send on same serial.

unitware, 2018-01-10:
cool, I'll try that in a few days when I'm feeling lucky.. Any instructions on how to do the upgrade somewhere? (I know, I'm lazy -  but its sometimes hard to find the most accurate instructions..)

qba667, 2018-01-10:
@unitware 
Description is here:
https://github.com/povlhp/FlySkyRxFirmware
Firmware with same serial:
https://github.com/qba667/MAVLinkToIbus/tree/master/IA6B

qba667, 2018-01-15:
@unitware 
I need small explanation. First of all I have implemented configurable IBUS telemetry in Cleanflight.
But I have few problems while testing using unit tests.
I am using preprocesor directives #ifdef to exclude functionality  when for example GPS or BARO is not compiled. But during linking I am getting error about undefined variables related to those features. 
So those seems to be defined because code gets compiled but variables are not visible.

unitware, 2018-01-15:
Is it the same after a clean build?

If so then BARO is defined and you will need to stub it. Which branch are you working in?

Skickat från min iPhone

qba667, 2018-01-15:
@unitware thanks I have did it already. Seems to be fine but I have to define many stubs in the test.
I am working on my own copy of main branch. After cleaning up of code I will create PR.

Yenya, 2020-05-28:
@unitware, @qba667 - this is not related to the Tx firmware, isn't it? Can I close the issue here?

lmanchanda75, 2021-07-21:
> cool, I'll try that in a few days when I'm feeling lucky.. Any instructions on how to do the upgrade somewhere? (I know, I'm lazy - but its sometimes hard to find the most accurate instructions..)

lmanchanda75, 2021-07-21:
> cool, I'll try that in a few days when I'm feeling lucky.. Any instructions on how to do the upgrade somewhere? (I know, I'm lazy - but its sometimes hard to find the most accurate instructions..)

@unitware  were you successful in this ?