qba667/FlySkyI6 #47: Binding/Arming issue

DrRats, 2018-01-15:
I updated my Flysky I6 with the latest firmware. Flyplus 1.75.  When I bind my transmitter to the receiver it says "binding ok" and the LED on the receiver is solid red. After this happens, I cannot arm my quad (I am using the eachine wizard x220). I've tried stick arming, switch arming, etc...When I am in betaflight and try to set up the receiver I cannot see any of the imputs move (Throttle,pitch, yaw, roll) (From 1000 to 2000) in the tab. It is like the transmitter is not communicating with the quad. Do I have to change any settings in betaflight to get the transmitter to communicate with the quad? My transmitter worked without issues before the firmware upgrade and haven't changed anything since the firmware upgrade. Please help!

qba667, 2018-01-16:
@DrRats if you were using much older version maybe there is change in RX mode. Search for such setting - I do not remember the name. There you can change output mode
IBUS/SBUS/PPM/PWM

DrRats, 2018-01-16:
I could not get it to work the way it was. I ended up changing the receiver port on the flight controller from ppm to ibus and changed the pin connection on the actual receiver to servo/ibus. After I changed my betaflight settings from ppm to ibus it worked fine. The input mode on the rx was on ibus/ppm. There was not a ppm only setting.

qba667, 2018-01-19:
@DrRats exactly because those modes are configured together. No idea what was wrong PPM output is working just fine with my receiver.

closed 2018-01-19 by qba667