qba667/FlySkyI6 #49: Sticks Mode not working

JensBoerner, 2018-01-21:
Having updated to latest release 1.7.5 (First with the CLI updater, then with the GUI) I am experiencing a problem with the startup. Throttle needs to be up, not down, it seems to be reversed. Also Whatever Stick Mode I choose, Channel 3 is always on the Throttle lever.

JensBoerner, 2018-01-21:
Update: In Mode2 it starts up with stick to the bottom, I noticed after a factory reset.

Btw. Would it be possible to add Channebl 7+ to the Mix and Reverse Menus?

qba667, 2018-01-21:
@JensBoerner Reverse for sure. But for mix hard to say because those methods are somewhere deep in original firmware.

qba667, 2018-01-21:
@JensBoerner maybe it will be possible to add those channels to mix as well - I will create test build.

qba667, 2018-03-13:
Working exactly as in stock firmware.

closed 2018-03-13 by qba667