qba667/FlySkyI6 #50: Afhds / afhds2 lock up

Whitey2505, 2018-01-25:
Excellent work on the firmware.
There's so much information in so many places that this may have been highlighted already if so my apologies.
Are you aware that when you try to change receiver type from AFHDS2 to AFHDS,
 as soon as you set AFHDS2 to off the TX locks up (no button selections register at all) and you have to turn the TX off and back on to unlock it?

Thanks again for the hard work 👍👏👏👏

qba667, 2018-01-25:
@Whitey2505 
I was not aware about it - thanks for info - I will try to check it. Is the value saved?

Whitey2505, 2018-01-25:
No, value isn't saved so you can't switch out of AFHDS2. You can't back out of the menu either. As soon as you set it to off the TX locks up.

qba667, 2018-01-25:
@Whitey2505 just verified against latest FW. I can not reproduce the issue. Can you provide more details about your setup?

Whitey2505, 2018-01-25:
Yeah, absolute standard TGY-I6.
I'll try it again to be certain but did it twice yesterday.

Whitey2505, 2018-01-25:
Really wierd, that worked fine this time. Put it down to user error I guess. Thanks again 👍

qba667, 2018-01-25:
Okey, if this occurs again let me know - we will trace it and fix it.

closed 2018-02-08 by qba667