qba667/FlySkyI6 #52: FS-CAT01 /Sensor List IA6B

JensBoerner, 2018-02-16:
Hello,

It's rather a question, then an issue: refering to https://github.com/benb0jangles/FlySky-i6-Mod-/issues/48 (sorry having mixed the respositories):

Using IA6, IA6b and IA10 receivers i noticed currently (Firmware 1.75 for FS-I6) I only get an altitude in the sensor list for IA10. Actually I get Temperature and Pressure from CAT01 and an caclucated Altitude, but over Sea Level. In the Extra menu I can either change Pressure or temperature or set it to the current measures one, but not e.g. reset altitude to 0 to get Altitude over Ground.
In IA6B I only get Pressure, Temperature is always set to 25°, and I do not get any altitude or temperature in the sensor list.
I aim for updating the IA6B Formware, but not completly sure what version to choose and if this really is the problem.
Any thoughts on that? Is the problem an outdated IA6B Firmware having not all sensor data? Voltage, RPM and Temperature (extra Sensor) works as soon as I plug it in.

Oh and BTW. when addeding Temperature Sensor and CAT01 Pressure Sensor It does not work, the sensor list is shortened to end after the data received from the frist sensor, probably because CAT01 Temperature sensor data interfers with the FS-CTM01 Temperature sensor.

qba667, 2018-02-17:
@JensBoerner The problem with IA6B is that is not supporting 4 bytes sensors - firmware 1.6 should fix it. Maybe altitude calculation is missing because of missing temperature.
Stock firmware in version 1.6 can be found here:
https://github.com/povlhp/FlySkyRxFirmware

If you set value of pressure and temperature to value provided by sensor you will get 0m of altitude measurement. 

With missing temperature sensor there is another problem - I have remove sensor index so 2 sensors with same index will not work we will always get last of sensors. Bu possibly in sensor display you should see it.

Yenya, 2020-05-12:
@JensBoerner, does the CAT01 work for you in 1.7.6? It should work now.

Using multiple sensors of the same type (e.g. two temperature sensors) is not supported, though. I recommend to fill this as a separate issue.

mazi14, 2022-10-15:
I have bought FS-CAT01 altitude sensor recently and I have got FW FlyPlus 1.7.1 installed in my FS-i6 transmitter. I am using FS-iA6B receivers. I can see temp, press and altitude in sensor list page, but values 216,8°C, 1465,19Pa and -4254,39m never change no matter of actual altitude or outside temperature (with Taranis X7 transmitter and multimodule I get the same resul with exception that altitude is reset to 0m automaticaly). I can adjust press or temp in ASL menu, but can not reach altitude 0m. Is it common behavior of this sensor? Will upgrade to FW 1.7.6 help?
[image]
[image]
[image]

dolie, 2022-10-31:
> @JensBoerner, does the CAT01 work for you in 1.7.6? It should work now.
> 
> Using multiple sensors of the same type (e.g. two temperature sensors) is not supported, though. I recommend to fill this as a separate issue.

Hello,
CAT01 does not work with flyplus 1.7.6 with (unupdated) FS-iA6B. It only returns pressure.