qba667/FlySkyI6 #55: Correct voltages readings for a 2S internal battery.

bsdshneg, 2018-03-05:
After the power improvements, namely the replacement of the stabilizers 6206a by ams1117 and the installation of the 2S lithium-ion battery, the maximum display voltage of the inner battery is equal to 6.65V. I understand that this reading of the values from the analog port is registered in the firmware. Tell me, is it possible to have another version of the firmware taking into account the work from the 2S lithium battery?
Thanks!

qba667, 2018-03-05:
@bsdshneg
Replacing stabilizers was not necessary.
What you need to do is replacing resistors R9 and R10 - creating voltage divider for analog input of MCU.
Voltage calibration can be made in factory menu (sticks moved to left down before switching on the unit).
I have used 4.1k as R10 and 10K as R9.
[image]

bsdshneg, 2018-03-06:
Thank you!
But stabilizers 6206 up to 6v and 2s li-ion is 8.4v.

qba667, 2018-03-06:
@bsdshneg 
you are correct - but it is strange that your unit contains 6206 - mine contains 9 V tolerant regulators (don't remember symbol).

bsdshneg, 2018-03-06:
Maybe different revision ...
Thank you for resistors nominals )))

Still a question, time speech about it has come. How did you put the 2S battery? What kind of battery was used?

qba667, 2018-03-07:
@bsdshneg 18650 after removing original battery holder will fit. 
[image]
[image]

closed 2018-03-13 by qba667