qba667/FlySkyI6 #56: Wish erase the start up check

Holger71, 2018-04-12:
Hello is it possible to erase the start up check and the modellselect protection?
I used the FS-i6 for Trucks, Tractors and so on. The Throttle Stick is mod. to middle position.
To change between to Modells without run on pacour to switching off/on the Modells would be fine.

qba667, 2018-04-12:
@Holger71 search in this topic:
https://www.rcgroups.com/forums/showthread.php?2486545-FlySky-FS-i6-8-channels-firmware-patch!/page174
I have provided once a version without checks.

StefanKellerAC, 2018-05-11:
sorry, but I cant find it.
( One Time I had an unused switch broken by transport and couldnt use the transmitter! )

Is there a possibility in 1.7.5. maybe selectable in the menu :-) ?

b.t.w:  lot of thanks!!!

qba667, 2018-05-11:
@StefanKellerAC  I will publish it here later.
Selection from menu only if we get some space - and free time:)

StefanKellerAC, 2018-05-11:
for me its not important to have the check, so a different compilation would please me :-)
is there a tutorial how to compile?
what do I have to change to disable the check?

Boberling, 2018-06-05:
Hi!
I have problem with "Warning Place all swithes  up"
I have alredy cut few millimeters of stoppers like in a video from youtube.
I checked with a multimeter all the swithes and swaped the sticks.

Tried few firmvares, original and not.

I could enter Factory meny to see wich stick or button in wrang position (lef and bottom sticl position would not work probably)

So i wish i could turn off this start scheck somehow.

m42uko, 2018-10-26:
Can someone point me in the right direction on what I need to change in the firmware to remove this check? I have (hopefully correctly) disassembled the firmware using radare2, and I am able to compile a new firmware from this repo. I just have trouble finding the check in the code.

So, if it's not too much work, can someone (maybe @qba667 as you've already done it once) point me to the offset (or part of the code) where this function is implemented.

Thank you!

m42uko, 2018-10-26:
Thanks. That worked.

(But then I tried to remove the stick-zero-check as well and bricked my remote in the process; only a blinky screen, and I can't get into the bootloader anymore; and my ST-link (STM32 dev board) doesn't seem to like to connect to this CPU either. I'm surprised to see that it is possible to brick the bootloader even from the serial upload-thingy...)

qba667, 2018-10-26:
@m42uko are you sure that you have calculated checksums correctly?
To connect to the MKL chip you need to make JLINK out of STLINK:
https://www.segger.com/products/debug-probes/j-link/models/other-j-links/st-link-on-board/
Then to write use old version of JFlash (JLinkARM_V486b).

m42uko, 2018-10-26:
I managed to get it back to life using the same method you described, and it was indeed the CPU hardfaulting. But setting up the J-Link, OpenOCD, and getting the chip programmed was an absolute nightmare... so many pitfalls.

To generate a new version, I modify `source/build/org.bin` und use `make` to build a new version, so the checksums should be alright.

Anyways, I changed two instructions. The one you described (that worked, but only for the switches) and the one I thought was for the sticks. But I guess I was wrong about that one. Here's the diff:
```
 Disassembly of section .data:
@@ -11742,10 +11742,10 @@
     6356:	0300      	lsls	r0, r0, #12
     6358:	4d48      	ldr	r5, [pc, #288]	; (0x647c)
     635a:	0f00      	lsrs	r0, r0, #28
-    635c:	d10d      	bne.n	0x637a
+    635c:	64c0      	str	r0, [r0, #76]	; 0x4c
     635e:	f7fa fd03 	bl	0xd68
     6362:	02c0      	lsls	r0, r0, #11
-    6364:	d509      	bpl.n	0x637a
+    6364:	64c0      	str	r0, [r0, #76]	; 0x4c
     6366:	4846      	ldr	r0, [pc, #280]	; (0x6480)
     6368:	7980      	ldrb	r0, [r0, #6]
     636a:	2801      	cmp	r0, #1
@@ -25487,8 +25487,8 @@
     d6c4:	2f30      	cmp	r7, #48	; 0x30
     d6c6:	3831      	subs	r0, #49	; 0x31
     d6c8:	3220      	adds	r2, #32
-    d6ca:	3a32      	subs	r2, #50	; 0x32
-    d6cc:	3934      	subs	r1, #52	; 0x34
+    d6ca:	3a30      	subs	r2, #48	; 0x30
+    d6cc:	3630      	adds	r6, #48	; 0x30
     d6ce:	0000      	movs	r0, r0
     d6d0:	ef10 0000 	vhadd.s16	d0, d0, d0
     d6d4:	f880 1fff 	strb.w	r1, [r0, #4095]	; 0xfff
@@ -27375,5 +27375,6 @@
     e750:	4c53      	ldr	r4, [pc, #332]	; (0xe8a0)
 	...
     e7fa:	0000      	movs	r0, r0
-    e7fc:	8700      	strh	r0, [r0, #56]	; 0x38
-	...
+    e7fc:	5200      	strh	r0, [r0, r0]
+    e7fe:	Address 0x000000000000e7fe is out of bounds.
+

```

And I noticed that my radare2 disassembly produces garbage... I guess I really need to dig out my IDA installation somewhere to get this done. Unless you also have the address handy to disable the sticks-check that is ;)

But that's a thing for tomorrow.

Thanks a lot for you help!

m42uko, 2018-10-27:
Okay, now with a proper disassembler (not just objdump), I managed to figure out what I needed to modify. There are a couple more lines that need to be changed in order to remove all checks (the one you described only disabled the check for two of the switches.)

Here's the diff:
```
 Disassembly of section .data:
@@ -13424,10 +13424,10 @@
     7b56:	0300      	lsls	r0, r0, #12
     7b58:	4d48      	ldr	r5, [pc, #288]	; (0x7c7c)
     7b5a:	0f00      	lsrs	r0, r0, #28
-    7b5c:	bf00      	nop
+    7b5c:	d10d      	bne.n	0x7b7a
     7b5e:	f7fa fd03 	bl	0x2568
     7b62:	02c0      	lsls	r0, r0, #11
-    7b64:	bf00      	nop
+    7b64:	d509      	bpl.n	0x7b7a
     7b66:	4846      	ldr	r0, [pc, #280]	; (0x7c80)
     7b68:	7980      	ldrb	r0, [r0, #6]
     7b6a:	2801      	cmp	r0, #1
@@ -13437,7 +13437,7 @@
     7b72:	4844      	ldr	r0, [pc, #272]	; (0x7c84)
     7b74:	68a9      	ldr	r1, [r5, #8]
     7b76:	4281      	cmp	r1, r0
-    7b78:	e07c      	b.n	0x7c74
+    7b78:	d87c      	bhi.n	0x7c74
     7b7a:	f7fa fd1b 	bl	0x25b4
     7b7e:	2300      	movs	r3, #0
     7b80:	461a      	mov	r2, r3
@@ -13497,7 +13497,7 @@
     7c02:	68a9      	ldr	r1, [r5, #8]
     7c04:	30f5      	adds	r0, #245	; 0xf5
     7c06:	4281      	cmp	r1, r0
-    7c08:	e034      	b.n	0x7c74
+    7c08:	d334      	bcc.n	0x7c74
     7c0a:	e7b6      	b.n	0x7b7a
     7c0c:	43e0      	mvns	r0, r4
     7c0e:	02c0      	lsls	r0, r0, #11
```
I'll attach the updater.bin and org.bin for anyone else to play with:
[fs-i6_no_startup_checks.zip](https://github.com/qba667/FlySkyI6/files/2521636/fs-i6_no_startup_checks.zip)

I might take a look at how to make this a Makefile switch or something so that it's easier to build in the future. Maybe like the special version for sw_e. But I'll have to figure out how you're doing that first ;)

Again, thanks a lot, @qba667. You were a great help! :)

dremugit, 2020-11-24:
First, tremendous kudos for all the work that's been put into this firmware. The i6 has gone from a mid-range cheapo TX to an ohmigod-what-can't-I-do-with it device.

I'm sure I'm in the minority, but add me as another vote for regularly distributing "no checks" version(s) of the firmware when you release. I don't do flying things (for me they are invariably crashing things!) but there are trucks, tanks, forklifts, etc, all of which have self-centering sticks and/or non-stock default switch positions.

I know it'd a pain for you, going from SwE and no-SwE to four permutations, but if you could build "no-check" versions when you build, it'd be much easier for folks on the other end than recompiling. I kinda-sorta know what I'm doing, and it took me several days just to get the toolchain installed (and the right versions, and removing some old versions of gnu make, and getting the envars right and PATH in the right order, etc etc) never mind doing all the patching mentioned above. I *think* I made a 1.76 with no checks, at least it works on my TX's, but I'd hate to go through that next time around 👍

Cobalt6700, 2021-03-24:
Another vote here for "no checks" version(s) - I use these TX's for all sorts of things now (because of this firmware!), this makes them my go-to set - especially with 14 CH over i-BUS. Currently building a 14 CH model railway point + signal control unit using one, and having the startup checks disabled would be great (I currently have to set all of the switches back to one place to turn on the TX). 

May have to give re-building the firmware a go myself - never done anything like it before though. @dremugit would there be any chance of me grabbing a file from yourself?

dremugit, 2021-03-24:
This is 1.7.6-with-no-checks that I built. Usual disclaimers: Use at your own risk, may cause spontaneous laughter in laboratory chickens, etc.

Cobalt6700, 2021-03-25:
I'm pretty new to github, I'm guessing there is meant to be a file attached to the email comment but there doesn't seem to be one? Either that or I'm being a total derp and have no idea how to download an attachment from github. 

To check - is your build of the swe or non-swe firmware?

dremugit, 2021-03-25:
Bugger. Attachments prolly get stripped from the D-list. Send email to dremu-at-yahoo direct.
-- A

Cobalt6700, 2021-03-26:
@dremugit - Legend - thanks for the firmware 👍