qba667/FlySkyI6 #57: dual rate and expo linked to two trim scale

AeroVigarista, 2018-05-03:
Hi dear, great job!

I flashed the 1.7.5 ppm 8ch for swe, vey nice fw, but the fsi6 have a old knowed bug on fly mode normal and sport (default swd), menu dual rate/expo., the related function is "linked" to trims values.

Scenary: takeoff with normal fly mode on swd, trim the aircraft, change swd to agressive dualrate and the plane lost the center position, trim again and when return to normal swd the trim position is not centered again, the trim value is linked to two stick scale, expo have this issue too...

Maybe using trims to setup subtrims will solve, or trim when clicked perform its operation based on values of sport mode and keep it when normal mode applied, absolute trim function will be apreciated because it flaws on elevon mixer too.

It is not your fault, the original tx have this bug, if you have the skills to do it better a lot of people will enjoy.

Thanks

qba667, 2018-05-03:
@AeroVigarista
Thank for detailed description - finally someone has explained it correctly!
Right now I am working on minimal version of OpenTX for I6X - prognoses are promising - I can not confirm it right now but MAYBE it will also fit I6. Then the problem will be solved. 
If not I will try to address the issue in free time.

AeroVigarista, 2018-05-03:
thanks a lot, opentx sounds good news too.