qba667/FlySkyI6 #58: Test range

macem, 2018-05-08:
Question if there is possible to support in test range button.

qba667, 2018-05-09:
@macem please describe in more details how it should work.

Wes-Leigh, 2018-05-09:
From what I can understand, I think he is looking for a "Test Range Mode" similar to the Taranis OpenTX. Where the transmitter's power is lowered to, I think, 1/30th of the output. Then you can calculate maximum distance that the RSSI value drops out. (ie. Range Test)

macem, 2018-05-09:
yes as @Wes-Leigh described above.

Wes-Leigh, 2018-05-09:
@qba667 the question is, Is there a way to control the output power of the transmitter? This will be the deciding factor.

qba667, 2018-05-09:
@Wes-Leigh it is possible - the init sequence of A7105 can be manipulated.

Wes-Leigh, 2018-05-09:
@qba667 from a quick browse through of opentx, seems like it is TELEMETRY_RSSI they using to display. https://github.com/opentx/opentx/blob/462f6244aa02d41b179ed4f6af8ad20e8862c354/radio/src/gui/128x64/model_setup.cpp#L1494 I'm looking to see what TELEMETRY_RSSI is https://github.com/opentx/opentx/blob/2b604a49ea64a7471ca107e2980299481538bffa/radio/src/telemetry/frsky.h#L371 so seems to be coming from RX side and not the TX side. I'll see what else there is but seems its coming from the SPort value as either the RX or the FC.

qba667, 2018-05-09:
@Wes-Leigh got the concept - for sure it must be signal strength seen by receiver, same as it is in our case. For RSSI output I am using SNR instead of RSSI.
If we measure SNR with full power, then reduce output power to minim and measure SNR again, then using  free-space path loss equation we can estimate the range.
However in my option it is more important to check if radio link is not broken for low signals.