qba667/FlySkyI6 #59: PPM 6 and 5

Tonymony1, 2018-05-21:
Dear Friend,

I am trying to use basically this project into the Flysky i6:
https://www.rcgroups.com/forums/showthread.php?1677559-DIY-Headtracker-%28Easy-build-No-drift-OpenSource%29

They use arduino and a giroscope to generate PPM 6 and PPM 5 into the remote controller, I know it is for Taranys remote controller, but you make me happy with your update for Flysky i6.

I am feeding th ppm in the back of the remote controller, and assigned to the aux channells and when I press display channels I can see how the giroscope move on channels but the signal is not comming out from the remote controller.

Basically I have plenty problems trying to bind the remote controller, could you help me please?

REgards,

JakubMichal, 2018-05-22:
I am using it too, without changes. You need only to set it up...
https://youtu.be/imq3KJGTqHI

Tonymony1, 2018-05-22:
COuld you help me?

I am basically doing the following steps:

I have tried to power the arduino and gyroscope from the remote controller pin out. But the remote controller does not supply enough power to the arduino, then I have decied to only run the ppm cable and the ground like screen cable.
I am powering the arduino directly from the power supply.

Well it is very weird because when I connect the remote controller, the receiver looks binded LED red continuous but I dont get any output from the receiver. AnyHelp?

I am going to downgrade the firmware and try if the binding problem persists. I have two different receivers then if there is something wrong is the compatibility with my firmware or the remote controller is bad.

Regards,

JakubMichal, 2018-05-22:
I am out since end of week. Next week i should be at home and we can try to solve it.

Yenya, 2020-05-28:
@Tonymony1, do you still see the problem with the PPM input in 1.7.6?