qba667/FlySkyI6 #62: Failsafe isn't applied to sbus

supermomme, 2018-06-26:
I use a modded FlySky I6 with a X6B use the ibus output for sbus.
It works all fine, but if i turn off the reciever the channels go all to "default" positions.
_Throttel: 850 (I don't know why)
Yaw,Pitch,Roll: 1500
The Aux channels freeze.

And i can't use rssi/snr for Failsafe, because the aux channels freeze.

With the PPM Port the failsafe is applied and works fine.

Can anyone help me? I like to fly with more than 8 Channels.

qba667, 2018-06-27:
@supermomme as far as I know that is bug in receiver - updating firmware of X6B may help. Fail safe should be working also on IBUS, have you tried that?

supermomme, 2018-06-27:
My Flight Controller doesn't accept ibus from the reciever. I don't know if my flight controller is bad or the reciever.

Where can i find the Firmware for the X6B?

supermomme, 2018-06-27:
Is it that? [povlhp/FlySkyRxFirmware](https://github.com/povlhp/FlySkyRxFirmware)

qba667, 2018-06-27:
@supermomme yes that is correct repository, but I am not able to confirm that newer FW will fix the issue.

supermomme, 2018-06-27:
I'll try that.

supermomme, 2018-07-03:
I think i flashed it now, but I don't know how to verify it.
Still no failsafe on sbus :/
Do you have any other suggestions for what i can do?

supermomme, 2018-07-10:
closed due to no response

closed 2018-07-10 by supermomme

qba667, 2018-07-10:
@supermomme try different firmware for example from IA6B. SBUS issues are known...

supermomme, 2018-07-10:
Yeah i tried it with the IA6B firmware and it works on ibus but not on sbus. Thanks a lot thats what i want.