qba667/FlySkyI6 #64: Aux = VRx*SWy

AndreBL, 2018-07-03:
Hi,

I upgraded my FS-I6 with this modded firmware and that is a great improvement over the original.

A feature request, if allowed here:

In airplanes with flaps, with other radios I usually have a potentiometer (like VRA or VRB) to adjust the amount of flap extension and a 2-pos switch to activate and deactivate the flaps. In the Aux Channels menu an example would look like:

Channel5     VrA*SwB            (an asterisk for multiplication or "and" logic)

Which means the output of CH5=VrA but only if switch SwB is activated. Otherwise CH5 defaults to its minimum endpoint (flaps fully retracted).

It would be great to have such feature in the next releases. Maybe the same function could be accomplished if the Mix page allowed a physical switch to toggle a Mix on/off.

Regards,
André

qba667, 2018-07-06:
@AndreBL they way to go seems to be mix configuration - but the number is mixes is very limited, maybe it would be possible to change number of mixes instead.

AndreBL, 2018-07-06:
More mixes would be great. A huge improvement in flexibility if the user can toggle a mix on/off through a switch.

Yenya, 2020-05-11:
@AndreBL: we cannot toggle mix on or off, but we can set Ch5 as the VrA value, provided that the two-position switch is in a given position (and to zero otherwise). See the solution I offered in a comment to issue #82. For your issue, it should be similar: Set up `Ch5` as `VrA`, `Ch6` as SwB. Then use one of the mixes to do the following:

```
Mix is    On
Master    Ch6
Slave     Ch5
Pos. mix  100%
Neg. mix  100%
Offset    -50%
```

Does the above solve your problem?

Yenya, 2020-05-22:
I have documented this in the Wiki, and I am closing this issue:
https://github.com/qba667/FlySkyI6/wiki/Setup-Disable-Channel

closed 2020-05-22 by Yenya