qba667/FlySkyI6 #67: Ok button hold bug, failsafe 5 second bug, and 20kb upload error bug

panoramastudio, 2018-09-01:
[image]
1)when i go to menu->system->extra->timer, and hold OK button it shows 0 channel witch is not correct channel and does not exist, i think we need to rewrite this function 
2)failsave  work only when the rx and tx works with each other more than 5 seconds, in original it works emidietly

3)many many times load stops here-
[image]

closed 2018-09-01 by panoramastudio

qba667, 2018-09-17:
@panoramastudio have you solved those issues? 
I think first one is because too generic implementation.
Fails safe must be checked, there should be no changes in this area.

panoramastudio, 2018-09-17:
After it radio have no any  functionality, it does not turn on, than I tried to flash it with USB->UART of Arduino MEGA, and it flashed, all ok!