qba667/FlySkyI6 #69: Tx Battery Status

tictac80, 2018-09-17:
It was noted that the TX BATTERY VOLTAGE on the display is not as intuitive as the original BATTERY SYMBOL. Since the symbol requires more space to program, I would suggest a BATTERY PERCENTAGE be displayed instead of the actual voltage.

liguo-git, 2018-10-22:
% more intuitive

qba667, 2018-10-22:
@tictac80, that can be done there is enough space in main screen method to do so.

DusanF, 2018-10-22:
I don't think it is a good idea. To display it correctly, you would need discharge curve for every battery type that is usable (Alcaline, Ni-XX, Li-XX) and calculate percentage from it

destarita, 2018-10-22:
9y99no9no

Yenya, 2020-05-11:
I think the current firmware is OK without the battery icon - the primary source of the information about the empty battery should be the battery alarm - people do not have time to look at the screen when flying. And for the exact value one would need to know the battery type and discharge curve, as @DusanF said. So I think the current state of things is OK.

closed 2020-05-11 by Yenya