qba667/FlySkyI6 #70: Trainer Student Mode

hgg2k, 2018-10-20:
I have upgraded my I6 to the latest firmware and everything works fine
except the Trainer mode which I cannot make it work.  I have two similar
radios with same firmware and using the following cable from banggood:

https://www.banggood.com/Flysky-i6-Trainer-Cable-For-FS-i6-FS-T6-Transmitter-p-986742.html?rmmds=buy&cur_warehouse=CN

Quad is bound to radio A.
Radio B is set to student mode and the S symbol appears on the display.
Radio A is set to trainer mode and a switch is assigned and shown as engaged when flipped.
Radio A does not display any Trainer icon.

With the above setup, the trainer mode switch does nothing, Only radio A
can control the quad.

Do you think that this is because of the firmware upgrade or am I doing something 
wrong?

Thanks!

qba667, 2018-11-22:
@hgg2k there is no T icon at all. But if state is shown as engaged it should be fine. Are you sure that connection is correct? Some of users successfully run such setup. Important is that both radios must have same firmware version - be sure that you are using 8CH_PPM.zip.

Yenya, 2020-05-28:
@hgg2k, did @qba667 answer your question sufficiently? I use the trainer/student cable myself, and it works for me in 1.7.6 (except for the Issue #81, the Ch1 value being off). Can you test it again with 1.7.6? Thanks!

hgg2k, 2020-05-29:
Before I update the firmware to both TXs, I want to double check the following.
In the cable I am using to connect student to trainer, pin PPM from Student is
connected to pin PPM to teacher and ground to ground.  RX and TX pins are
not connected.  Do I need TX & RX for student/trainer communication?  Thanks.

Yenya, 2020-06-01:
No, the upper two pins (Rx and Tx) are needed for firmware update only. For student-teacher you need only two wires - GND, and PPM in from one side connected to PPM out on the other side. If you want to keep things simple, use also the third wire connecting PPM out from one side to PPM in on the other side. This way you don't have to worry about which end is supposed to be teacher and which is the student, it can be plugged in both ways.

Yenya, 2020-11-11:
This has been without response for several months now, I am closing this issue.

closed 2020-11-11 by Yenya