qba667/FlySkyI6 #71: Can I put the battery image back?

Yudgin, 2018-11-17:
Hello
I have a question. Can I put the battery image back?

qba667, 2018-11-22:
@Yudgin - is is already discussed - now it is a little bit more complex because of different batteries but should be possible.

Yudgin, 2018-11-22:
I use translator.
I do not understand, tried to draw.
The compiler writes "overlaps section .mod_code_65C8_68A7"
Not enough memory?
Can I remove the timers setting? This will help me?
I need a global array. Where can I post?
How much will it cost to complete work on i6x?

qba667, 2018-11-28:
@Yudgin 
You have to be careful where you adding your code. Removing some features helps a lot, you can also remove menus that you are not using. 
Global array in RAM or FLASH? 
Do you want to help with I6X I would say it is 2 months to get initial versions working.
Later I will provide you addresses of uart methods.

Yudgin, 2018-11-28:
I'm nice.
Global array in RAM.
I do not understand these matters.
I want to display the coordinates on the UART.
I want to keep a log of coordinates.
Thank.
How can I help I6X?

Yudgin, 2018-12-18:
Hey. I remind of myself. "Later I will provide you addresses of uart methods."

qba667, 2018-12-20:
@Yudgin 
Here you have where, UART 2 is configured and  DMA is being initialized (0x978):
`
signed int setupUART2()
{
  signed int result; // r0

  MEMORY[0x20000038] = 0x20000800;
  MEMORY[0x2000003C] = 0x20000800;
  MEMORY[0x20000040] = 0x20000800;
  MEMORY[0x40048034] |= 0x1000u;                // uart 2 enable
  MEMORY[0x4004D040] = 768;                     // PORTE_PCR16
  MEMORY[0x4004D044] = 768;                     // PORTE_PCR17
  MEMORY[0x4006C000] = 0;                       //  UART Baud Rate Register: High (UART2_BDH)
  MEMORY[0x4006C001] = 13;                      // UART Baud Rate Register: Low (UART2_BDL)
  MEMORY[0x4006C003] |= 0xACu;                  // UART Control Register 2 (UART2_C2) 
  MEMORY[0x4006C008] |= 0xA0u;                  // UART Control Register 4 (UART2_C4)
  MEMORY[0x40021000] = 0x87u;                   // DMA channel mutiplexer 0 
  result = 1073774848;
  MEMORY[0x40008104] = 0x4006C007;              // Destination Address Register (DMA_DAR0
  MEMORY[0x4000810C] = 0x20520080;              // DMA Control Register (DMA_DCR0)
  MEMORY[0x40021001] = 0x86u;
  MEMORY[0x40008110] = 0x4006C007;              // Channel Configuration register (DMAMUX0_CHCFG1)
  MEMORY[0x40008114] = 0x20000800;              //  Destination Address Register (DMA_DAR1)
  MEMORY[0x40008118] = byte_FFFF;               // DMA Status Register / Byte Count Register (DMA_DSR_BCR1)
  MEMORY[0x4000811C] = 0x601A0800;              // DMA Control Register (DMA_DCR1)DMA Control Register (DMA_DCR1)
  return result;
`
We have to check if DMA is not used for anything else when firmware is running. Memory adres must be also verified.

Yudgin, 2019-01-11:
Congratulations on holidays.
I honestly did not understand what to do.

Yenya, 2020-05-28:
@Yudgin, do you still need to have the battery image? I think the voltage reading + the audible alarm which 1.7.6 has provides sufficient feedback on the battery state.

Yenya, 2020-11-11:
This has been without response for several months now, I am closing this issue.

closed 2020-11-11 by Yenya