qba667/FlySkyI6 #72: FS-IA10 using PWM

mukhdi, 2018-11-20:
Hi qba667 ... im very happy with your work ... I use IA10 rx hoping can get clean 10 channel for servo not for switches ... im not finding a way to use all my 10 servo in pwm mode ... channel 7 to 14 are swithces ... what am I missing ... thanks

mukhdi, 2018-11-22:
Anyone

qba667, 2018-11-22:
@mukhdi please provide more details - how are you distinguishing servo from "switches". Reciever output should be correct because it is not able to distinguish at all.
The unit has ADC converters used for SWB and SWC, VARA and VARB it should be sufficient to output all 10 channels in full range, so it should be sufficient to output all PWM signals on IA10.

mukhdi, 2018-11-23:
@qba667 Thanks for ur reply
Im building a foamy plane with 4 motor without using any flight controller, so the scheme like
Ch1 for aileron left
Ch2 for elevator
Ch3 for throttle motor 1
Ch4 for rudder
Ch5 for aileron right
Ch6 for throttle motor 2
Ch7 for throttle motor 3
Ch8 for throttle motor 4
Ch9 for flap right
Ch10 for flap left
I can use Ch9 and Ch10 for flaps with switches but i cant find a way to use ch7 and ch8 to mix with ch3 .. in the mix menu, the selection is only until ch6

Thanks for your help

qba667, 2018-11-28:
@mukhdi Now I go the point, many people are asking me about more mixes. To implement it I have to first understand how it is working in oryginal firmware, then find the implementation and rewrite it.
I am working now on OpenTx for I6X maybe as side effect we will get something for I6.

mukhdi, 2018-11-29:
@qba667 this is really a good news and I believe the community also waiting for this ... just a suggestion would you mind to open a donation paypal account to the community in case there is a member who wants to give appreciation of your work ... Cheers

qba667, 2018-11-29:
@mukhdi - it will take me probably whole winter and then fitting into I6 next few months - maybe meantime I can provide a temporary solution - maybe I can expose channel 3 (throttle) on channel assignment page so you can set it up.

mukhdi, 2018-11-29:
@qba667  awesome ... thanks mate

mukhdi, 2018-12-13:
Hi Jakub

Hope the fw is already to hit my FSI6 

Cheers

qba667, 2018-12-13:
@mukhdi yes it is, together with few fixes of other issues. You will get custom build probably.

mukhdi, 2018-12-13:
Great ... how much will it charge my friend .. thanks

Ethanator10000, 2019-04-11:
Does your firmware add the extra channels over PWM too? I thought that it only added them over PPM or S/Ibus but now I'm confused. Could I bind to a ia10 RX and plug in 10 servos just like that? (Plus a power source of course)

Thank you for all the work you have done. I have been following this project for a while now and I think it's absolutely amazing what you have accomplished.