qba667/FlySkyI6 #73: Adjust 6-channel menu items to full 10-channel

Saijin-Naib, 2019-01-03:
A few critical system menus are still six-channel, despite the firmware & the display screen reflecting all 10 channels.

Menus still 6-channel limited:
Reverse
End Points
Subtrim

This would be incredibly helpful for setting up auxiliary function switches, such as SwD as my beacon toggle for betaflight, which expects inversed signal to enable the beacon, which I can't do since SwD is channel 8.

Thanks for the amazing firmware!

Saijin-Naib, 2019-01-04:
Stupid question, but if you changed the Settings-> Subtrim / EndPoints menus to support 10-Channel and took out the menus in the Extra area, would that no free up some space?

Similarly, would changing the Reverse screen from using graphical toggles to Y/N or something else free up any appreciable space?

qba667, 2019-02-05:
@Saijin-Naib It could be done, but method becomes more complex usually after changed - and it is hard to fit it on the place of old one.

Dredan72, 2019-02-25:
How do I silence the the alarm after flashing the 1.7.5 final with 8 channel ppm. The alarm goes off immediately after flashing complete.

Cobalt6700, 2020-08-07:
Is the reverse function for channels 6+ a possibility? I would really like to change the direction on CH6-10 for a project I'm working on.