qba667/FlySkyI6 #75: Model memory 13,14,15 issue

leeit2me, 2019-02-18:
Model memory 13 has a voltage bug that register the RX @ 8.7 volts which should register 5volts +/- . all other models are ready correct voltage. 

Model memory 14,15 the timer isn't saving. Once you save the setting and power off and you go back to the timer menu you will see all 0's in the setting fields in the timer. (model 14). The other model 15 has a set #'s but can't delete it as it will show again once you power cycle the tx.  

could this be a bug in this firmware?  Tested with another i6 tx and it mimic the same issue.

qba667, 2020-05-02:
Fixed in 1.7.6 will be released soon.

Yenya, 2020-05-09:
@leeit2me: 1.7.6 has just been released, with a fix to this issue by @qba667. I am closing the issue now. If the problems persist in 1.7.6, please reopen.

closed 2020-05-09 by Yenya