qba667/FlySkyI6 #79: Latency, Filesafe

HarutyunGazyan, 2019-05-17:
Hi, do the latency of this firmware is bigger than the original one?
And about filesafe, after turning radio on,  it need about  10 seconds to make filesafe work, so if i turn radio on, and I change stick position, and turn off the radio, receiver do not get filesafe, and servo continues to hold previous stick position, if I turn on radio, change stick position, and wait about 10 sec, and turn off, receiver gets the filesafe, and servo goes into filesafe default position, and do not hold old stick position as it happens in the first case.

Yenya, 2020-05-28:
@HarutyunGazyan, I am not aware of this firmware having a bigger latency than the stock one.

As for the failsafe - could you test it again with 1.7.6?

Yenya, 2020-11-11:
It has been NEEDINFO for several months now without response, so I am closing this. As I read the original report, it could be related to #94.

closed 2020-11-11 by Yenya