qba667/FlySkyI6 #81: In Trainer Mode，PPM input ,ch1 get value is notcorrect

betterlzm, 2019-06-26:
Excuse me，
One FS-IA6B(Named RX-B) ppm out signal connect to pad which is marked with ppm-in on the PCBA of FS-I6(Named TR-A).
The FS-I6 (Named TR-A) is in Trainer Mode, and send signal to FS-IA6B(Named RX-A). but tch1'value is not correct in the FS-IA6B(Named TR-A) .IN display view ,the ch1' bar is blink in the FS-I6 (Named TR-A)  .FS-IA6B(Named RX-B) ppm out signal is OK in  oscilloscope.

See attachment.

[image]
[image]
[image]

[image]

plehong, 2019-06-26:
Hello
I had the same problem when I connected a stock FS-I6X as a student to my FS-I6 with a trainer cable. The CH1 position was fuzzy in the display screen and the servo position had lots of jitter.
Maybe the modded firmware uses some memory location used by the PPM trainer input ?

qba667, 2019-06-26:
Have you guys tried 8 channel PPM Firmware? https://github.com/qba667/FlySkyI6/releases/download/1.7.5/8CH_PPM.zip

betterlzm, 2019-06-27:
hello，Jakub，
The PPM is 8channels on OSC,I will try  8 channels PPM Firmware,Thak you!

Yenya, 2020-05-11:
I can confirm the problem: Ch1 is way off, even when connected directly from another FS-i6 over the mini-DIN cable (both radios with FlyPlus 1.7.6). I have "fixed" this by setting the "Student" FS-i6 aileron trim about half-way to the left (no dr/expo set on the Student Tx).

Unlike @plehong, I don't see any jitter there, though.