qba667/FlySkyI6 #82: Feature Request:  Modify "Mix" so that a channels input can be held at zero.

12earth, 2019-06-30:
I am using this controller for a Motor On Axle surface vehicle with 2 ESC's.  

I would like to be able to set Aux Switches to have a "mix mode" that disables input for a given channel. (e.g. When Aux Switch 5 is thrown, zero volts is delivered to channel 1 which is the throttle for my front axle.  The effect of this would allow me to "lock up" my front or rear axle to navigate obstacles better.)

Is this possible in the current configuration?

Yenya, 2020-05-11:
@12earth, the mixes are too hardwired deeply into the original FlySky code, which we (FlyPlus firmware) did not touch yet. However, I think what you want can be done as an ordinary mix. Let's say you have `SwA` set up as `Channel 5`. Then go to the mixes menu, and set up one of the mixes the following way:

```
Mix is    On
Master    Ch5
Slave     Ch1
Pos. mix  100%
Neg. mix  100%
Offset    -50%
```
You can verify that this does what you want in the `Key` ⭢ `Display` menu.

Yenya, 2020-05-22:
I have documented this in the Wiki, and I am going to close this issue:
https://github.com/qba667/FlySkyI6/wiki/Setup-Disable-Channel

closed 2020-05-22 by Yenya