qba667/FlySkyI6 #84: PPM output (Trainer port) only 6 channels

mpm1396, 2019-11-18:
Hello guys, absolutely amazing work, so awesome. I only have one small problem or a feature request. So I am using my fs-i6 with a r9m in ppm mode. I am using the ppm output from the solder point on the i6 but basically it should be the same rail as on the trainer port. Problem is I only get 6 ppm channels out not all of my 10. Would it be possible to add the other 2 channels to the output as well? Thanks in advance.

gugpigto, 2020-01-11:
I've got the same problem. Have the r9m connected and it's working fine, but there's only 6 channels. 

Is there a way to get more PPM output channels?

mpm1396, 2020-01-11:
@gugpigto  i haven't found a way. But I switched radio lately. If you find a solution please keep me updated:)

deividfei, 2020-01-28:
I've got the same problem too. I try use a PPM4, 5 and 6 with mix a other channel, but the menu not has this possibilit. Is limited a VRA and VRB...

deividfei, 2020-01-29:
With the "8ch ppm" archive in the 1.7.5 version, work 8ch in the trainer Port! Very good! It's possible insert more channels?

qba667, 2020-04-29:
@mpm1396 @mpm1396 it will require to much changes - maybe one day it will be completely rewritten!

Sekilsgs2, 2020-08-13:
Hi qba667
What about this? How increase ppm channels for trainer port? Or maybe we can make sbus out for this pin?? Or sbus out with tx pin ??