qba667/FlySkyI6 #87: Random values on model 13,14,15 from ch5 to the rest

Gonnir, 2020-02-24:
I use an Eachine i6 (rebranded Flysky i6) with Flyplus 1.7.5.  I cant use those model memories just for 4 ch models because the values of the other channels are random values and barely effected by the switches i assign to them.

1074748905, 2020-04-07:
i have the same problem on fs-i6

1074748905, 2020-04-07:
and endpoint can not be saved in model 13 14 15

1074748905, 2020-04-09:
in model 13,if you change endpoints values, it will cause wrong ch1~ch8 values in other model.And then TX will not start (power on check) .

1074748905, 2020-04-09:
@qba667 can you fix the bug above i said？

qba667, 2020-04-29:
@1074748905 I just found what causing bug with models 13-15, in next build it will be fixed.

qba667, 2020-05-02:
@1074748905  can you verify change with attached file
[fs-i6_updater_05_02_15_23.zip](https://github.com/qba667/FlySkyI6/files/4567799/fs-i6_updater_05_02_15_23.zip)

1074748905, 2020-05-03:
> @1074748905 can you verify change with attached file
> [fs-i6_updater_05_02_15_23.zip](https://github.com/qba667/FlySkyI6/files/4567799/fs-i6_updater_05_02_15_23.zip)

@qba667 you deleate model 15,, only 1~14?     after testing model 13and 14 ，i find there is no error。thank you！

1074748905, 2020-05-03:
qba667 i  find a bigger bug，that bug cause great lag in pwm mode，do you see it？
https://github.com/qba667/FlySkyI6/issues/94

Yenya, 2020-05-09:
@1074748905: see the discussion at #94.

As for model 13 to 15 data overwriting, I consider it fixed in 1.7.6. Should the problem persist, please reopen this issue.

closed 2020-05-09 by Yenya