qba667/FlySkyI6 #93: "Extra" menu items missing for AFHDS models.

marktbaldridge, 2020-04-29:
If a model is AFHDS not AFHDS 2A, some of the "Extra" menu items are missing. Only the Timer exists. I can workaround by switching to an AFHDS 2A model, changing the "Tx Bat" settings for instance, and then switching back. But this should be fixed. Not sure if this project is alive any more. Would be interested in contributing.
[image]

qba667, 2020-04-29:
There is bugfix version release coming soon - maybe I can address it as well.

marktbaldridge, 2020-04-29:
Wonderful! I'm glad the project is still alive! I've made a small donation. :)

qba667, 2020-04-29:
@marktbaldridge thanks, it was not necessary, during coming weekend I want to address few topic on this old project. Stay tuned - there is no much space for changes but some fixes will be done:)

qba667, 2020-05-02:
[fs-i6_updater_05_02_15_23.zip](https://github.com/qba667/FlySkyI6/files/4567783/fs-i6_updater_05_02_15_23.zip)
Please try this one out.

marktbaldridge, 2020-05-06:
That fixed it! Thank you so much!! I believe I will be using this firmware permanently now!

closed 2020-05-06 by marktbaldridge