qba667/FlySkyI6 #94: great latency to control servos

1074748905, 2020-04-30:
I use 1.7.5 FW TX and iA6b RX. it will sometimes happen  when you turn on RX first and then TX.At the same time，home screen display intv ，rssi， err， SNR are not displayd

Yenya, 2020-05-05:
This happens to me as well sometimes, it can be fixed by powering the Rx down and up again. As such, I think it is caused by a power-on glitch, and not by the Tx firmware. I *think* (but I may be wrong) that it has happened for me even before I used the FlyPlus FW.

The symptoms are very sluggish servo response, no telemetry, and probably happens with ia6B only.

Yenya, 2020-05-11:
I consider this to be a brown-out issue of FS-iA6B, so I will close this issue. If you have any clues that it is a FlyPlus issue (e.g. comparing with the stock firmware), please describe it further here and re-open the issue. Thanks!

1074748905, 2020-05-11:
stock firmware never has this ploblem

Yenya, 2020-05-11:
OK, reopening.

Yenya, 2020-05-12:
Yesterday it happened for me again with my Sky Surfer 1400 (FS-iA6B on-board). If I remember correctly, it happened when I powered up the plane first, then powered up the Tx with a different model selected, and then switched to this model.

Another interesting thing is that the problem went away by itself after about 30 seconds without needing to disconnect the on-board battery and reconnect it back. After that, I spent a good one hour sloping with no control problems at all.

Then I went on to simulate this problem on a spare Rx I have, but was not able to reproduce it. Any hints how to reliably reproduce this problem are welcome.

1074748905, 2020-05-12:
> Yesterday it happened for me again with my Sky Surfer 1400 (FS-iA6B on-board). If I remember correctly, it happened when I powered up the plane first, then powered up the Tx with a different model [...]

power on FS-i6 Tx first  and then FS-iA6B。if there is  no bug， try to  power down FS-iA6B for  1~3s，and power on it  again 。   there is about  probability of 40%~50% getting this bug

Yenya, 2020-05-12:
I am still not able to reproduce this.

I have discovered that I had a flaky connection from my BEC to the Rx in my Sky Surfer 1400. After resoldering the connector, I tried to plug the battery in and out for at least 10 times with different power-down times in between, even with short power-up times, but the telemetry was immediately available and servos were resposive every single time.

I have also tried this with my spare iA6B, with one servo connected, and was also not able to reproduce this (tried also at least 10 times). Strange.

Do you have an un-modified Tx with stock firmware available for testing? If it happens to you, does the problem go away after a minute or so?

1074748905, 2020-05-12:
> Do you have an un-modified Tx with stock firmware available for testing?

Do you have an un-modified Tx with stock firmware available for testing?-  i Flash back to stock firmware ，bug never appear.

with 1.7.5 FW,  bug  sometimes   go away after several minutes