qba667/FlySkyI6 #97: Possible issue with iBus telemetry using iNav.

KiwiDavid, 2020-05-11:
Hardware: 
FS-i6 tx using upgrade v1.7.5 or v1.7.6   (both show the same issue)
FS-iA6B rx
SPF3 FC running iNav 2.0.0

Using the ibus telemetry in iNav with the one wire (resistor/diode) connection to the FS-iA6B the YAW value returned is correct from 0 degrees to around 327.60 degrees then further rotation produces incorrect -ve numbers.
It appears as if the YAW angle (in 1/100s of a degree) is hitting a signed integer limit (32768).  One of the iNav developers has indicated that the FC uses an unsigned integer for this parameter so the problem may be the way the FS-i6 is handling the data fed to it.

Best regards,
David

Yenya, 2020-05-11:
@KiwiDavid, you are right, FlyPlus considers yaw being a signed integer:

https://github.com/qba667/FlySkyI6/blob/master/source/source/mod.h#L291

Since I don't use iNav flight controller, I don't know whether it is correct or not: I would expect the Yaw being from -180° to 180° as a signed integer (and the same for pitch and roll).

Can you verify that this firmware fixes the issue?
[yaw-fix-test.zip](https://github.com/qba667/FlySkyI6/files/4609835/yaw-fix-test.zip)

If so, can you tell me which of the other sensors, as described in mod.h above, have a different signed-ness than iNav has?

-Yenya

KiwiDavid, 2020-05-11:
Hi Yenya,
Wow that was fast!
OK - looking at the list of variables anything that has a nominal value 
around zero and a deviation either side of that should be signed - e.g. 
Pitch and Roll.
Yaw is an arbitrary value between 0 and 359 degrees and is not centred 
around 180 degrees and therefore should be unsigned.
CoG is the same and should be an unsigned variable.
Heading is shown correctly as an unsigned value. To be honest I'm not 
clear on the difference between Heading and Yaw but Heading on my set up 
remains at 0 degrees and never changes.  CoG changes but is dynamic and 
derived from the GPS carrier Doppler so the craft must be moving.
So Yaw and CoG are the only variable I can spot that need changing.

I can check the code patch you have kindly linked to but I see it's for 
a tx with the Switch E mod.  Can I safely use that on my unmodified tx?  
I don't want a floating input pin.....

Best regards,
David

KiwiDavid, 2020-05-11:
Hi Yenya,
I just tried the new firmware but the same problem exists ( I hope I did 
it right).
Looking at the date of the file it would appear to pre-date me raising 
the issue so I'm wondering if it may not be the file you intended to send.
Let me know if  can help with any testing.

Best regards,
David

Yenya, 2020-05-12:
@KiwiDavid, you were right, I uploaded the wrong FW version. Sorry for wasting your time. Try this one instead:
[yaw-fix-test.zip](https://github.com/qba667/FlySkyI6/files/4614073/yaw-fix-test.zip)

KiwiDavid, 2020-05-12:
Hi Yenya,
Many thanks - that sorted it.
Yaw looks good now and is incrementing up to 359 degrees - no strange 
-ve numbers any more.
CoG also looks good but I couldn't do a robust test as I can't fly 
outside in the dark but by running with the craft it appeared to be 
going past the dreaded 327 limit with no problem.  I hadn't actually 
checked this on screen before but the code listing indicated a signed 
variable and I believe it should be unsigned.
Many thanks for your efforts in providing a fix for this.   If I can 
help in any way please contact me.

Best regards,
David

Yenya, 2020-05-12:
@KiwiDavid: I have modified only Yaw, not CoG in the test build attached to my previous comment.

Thanks for testing this. Now I changed both of these sensors to be unsigned in the source code (commit #69d0035fab643d5bc29f8be4be3bc6d5c18f00c3), so the fix should also be in the next FlyPlus release.

-Yenya

closed 2020-05-12 by Yenya

KiwiDavid, 2020-05-12:
Hi Yenya,
Thanks for the notification.
The reason CoG is working without change is because it is only using the 
nearest degree whereas Yaw is using 1/00 degree resolution (somewhat 
pointless).
However CoG is still never going to be -ve so really should be unsigned.

Best regards,
David