qba667/FlySkyI6 #100: Flashing a new chip (including the bootloader)

DeADLoCK-GUI, 2020-05-16:
i accident had set my ftdi adapter to 5v and blew my fsi6's RX pin so i found the IC -MKL16Z64VLH4 for cheap on aliexpress and i can easily solder it on the mainboard but how can i flash the new IC including the bootloader. If somebody knows upload a few pictures and the pin diagram of the programmer to the fsi6 mainboard

Yenya, 2020-05-18:
@DeADLoCK-GUI, I have never used it myself, but it should be possible to upload the firmware including the bootloader using JTAG interface. Some resources are here:

https://github.com/ThomHPL/KL16-FT2232-OpenOCD/
https://basejunction.wordpress.com/2015/09/13/en-flysky-i6-part-3-firmware-patching/

DeADLoCK-GUI, 2020-05-18:
> @DeADLoCK-GUI, I have never used it myself, but it should be possible to upload the firmware including the bootloader using JTAG interface. Some resources are here:
> 
> https://github.com/ThomHPL/KL16-FT2232-OpenOCD/
> https://basejunction.wordpress.com/2015/09/13/en-flysky-i6-part-3-firmware-patching/

so does the swe bin from the flysky mod github include bootloader so that i can flash the swe bin using stlink

Yenya, 2020-05-18:
The binaries from the Releases page (*_updater.bin, 59392 bytes) do not contain the bootloader. However, during the firmware build, the file named *_full.bin (65536 bytes) gets created. If I understand it correctly, the difference (0x1800 bytes) is the bootloader.

Attached is the fs-i6-swe_full_05_07_12_30.bin corresponding to the 1.7.6-swe_updater.bin. Hope this helps. If it does, may I kindly ask you to write a wiki page about this topic? What programmer did you use, and how did you flash the bootloader? Thanks,

-Yenya
[fs-i6-swe_full_05_07_12_30.zip](https://github.com/qba667/FlySkyI6/files/4644250/fs-i6-swe_full_05_07_12_30.zip)

ffastym, 2025-10-27:
> The binaries from the Releases page (*_updater.bin, 59392 bytes) do not contain the bootloader. However, during the firmware build, the file named *_full.bin (65536 bytes) gets created. If I [...]

How did you do that? The attached binary was flashed successfully and is working well, but my compiled builds have just a blinking display...